seemoo-lab/nexmon_csi #148: One problem about the CSI extractor

18563407351, 2020-11-24:
I  noticed that when collecting CSI, the raspberry pi cannot connect to wi-fi. So how does it realize the collection of CSI without connecting to WI-FI? And does this mean that the raspberry pi cannot send data packets to another raspberry pi through Wi-Fi during collecting CSI in the wireless channel?

zeroby0, 2020-11-24:
That's correct.

Your raspberry pi does not connect to any WiFi. Instead, it just listens to **all** packets going in the [channel](https://en.wikipedia.org/wiki/List_of_WLAN_channels) you configured.

It's very similar to monitor mode; instead of collecting packets, we collect CSI data. And yes, the Raspberry Pi cannot send data packets through the inbuilt WiFi chip. You can use a WiFi adapter on your raspberry pi to send wifi packets (not recommended. See #42) or use another Raspberry Pi or a different device.

Read this paper for a deeper understanding of how the tool works: [Free your CSI](https://dl.acm.org/doi/pdf/10.1145/3349623.3355477?casa_token=rMFDurPASXoAAAAA:hyM5WZXoxD8CV7WWG1OGiIbl15zIWint5OT4RCOTcETGY7j389b-Jr28ZaSnFjUdA7D6xEfRQDiH)

Mweme, 2021-09-01:
Hello Team @zeroby0 and others.
i'm currently extracting CSI signal basing on the good contributions you provide here, but when I type the command **sudo tcpdump -i wlan0 dst host 192.168.1.6 -vv -w data10.pcap -c 1000** it doesn't provide any response.

The last time I typed **sudo ifconfig wlan0 up** and I received the message **SIOCSIFFLAGS: Operation not possible due to RF-kill**
I tried to run the command sudo tcpdump -D, I get the following response
**1.eth0 [Up, Running]
2.any (Pseudo-device that captures on all interfaces) [Up, Running]
3.lo [Up, Running, Loopback]
4.bluetooth0 (Bluetooth adapter number 0)
5.nflog (Linux netfilter log (NFLOG) interface)
6.nfqueue (Linux netfilter queue (NFQUEUE) interface)
7.usbmon1 (USB bus number 1)**
However, the configuration procedures and installation was made successfully.

Any assistance will be highly recognized.