seemoo-lab/nexmon_csi #149: Utilizing this tool on Rasp pi 4B, no csi packet obtained

memory-xm, 2020-11-24:
greetings,

I am following the instructions to extract CSI on Rasp pi 4B.
Everything works well, i pass all step in Getting Started bcm4339 , while in final step, i find tcpdump cannot capture any packages after running long time (and I ctrl+c to manually stop it).

root@hammerhead:/data/local/tmp # tcpdump -i wlan0 dst port 5500 -vv
tcpdump: listening on wlan0, link-type EN10MB (Ethernet), capture size 262144 bytes
^C
0 packets captured
0 packets received by filter
0 packets dropped by kernel
Additionally information:

the phone will lost connection to a specific AP in monitor mode, after run command
ifconfig wlan0 up
nexutil -Iwlan0 -s500 -b -l34 -vm+IBEQGIAgAAESIzRFWqu6q7qrsAAAAAAAAAAAAAAAAAAA==
ifconfig mon0 up
if I skip the dst port, like tcpdump -i wlan0, I can successfully capture a lot of packets, however, these packets do NOT look valid, and /nexmon_csi/utils/matlab.m will tell error skipped frame with incorrect size, which means its WRONG to skip dst port.

nexutil -k
chanspec: 0x1001, 1

zeroby0, 2020-11-24:
Raspberry Pi has bcm43455c0, not 4339. I have a script to automate the install here: https://github.com/zeroby0/nexmon_csi/tree/pi-4.19.97

> the phone will l

Are you running it on a phone or a raspberry pi?

Yes, the connection loss is expected. You should not skip dst port, CSI packets arrive on that port. If you skip the port, you're capturing all the data. Most likely, your ssh packets. 

Which channel are you planning to collect CSI from? The chanspec looks odd, unless you're collecting from [2.4 GHz channel 1](https://en.wikipedia.org/wiki/List_of_WLAN_channels).

memory-xm, 2020-11-25:
Thank you for your reply, I am running the program on the Raspberry Pi, i pass all step in Getting Started bcm43455c0.
At the same time, I did not skip the port, the command used was: tcpdump -i wlan0 dst port 5500.
I check the channel of wlan0 through iwlist channel and no 157. I added the channel 157 by modifying the /etc/network/interfaces file and adding the following code:
auto wlan0
iface wlan0 inet static
address 10.3.141.1
netmask 255.255.255.0
network 10.3.141.0
broadcast 10.3.141.255
wireless-channel 157

My experiment is to send data packets to the Raspberry Pi through other devices to obtain the CSI in the data packets.

zeroby0, 2020-11-25:
Ah, yeah, I see where the problem is.

The easiest fix is to change your router to work in channel 36, and collect in channel 36.  `wpa_supplicant` is removed, so I think `interfaces` file may not be used.

Only some channels, as defined in this file: [src/regulations.c](https://github.com/seemoo-lab/nexmon_csi/blob/master/src/regulations.c), are used by the firmware. If you're operating in a channel without defining it there, no CSI will be collected.

Even though 157 is defined in regulations.c, maybe the control channel it is looking for is not 157. To add a new channel, first look at the list of 5GHz channels here: https://en.wikipedia.org/wiki/List_of_WLAN_channels

[image]

Channels from 149 to 161 are in one 80 MHz block. First set your router to operate in channel 149. This will be the control channel.

then add `CH80MHZ_CHSPEC(155, WL_CHANSPEC_CTL_SB_LL),` to regulations.c and recompile and install the new firmware.
`WL_CHANSPEC_CTL_SB_LL` is lower lower. If your WiFi is on channel `153`, then use `WL_CHANSPEC_CTL_SB_L`. Similarly `WL_CHANSPEC_CTL_SB_U` and `WL_CHANSPEC_CTL_SB_UU`

Even though the control channel is in 149, if your router is in 80 MHz mode, it will be able to use all of the 80 MHz from 149 to 161. But do note that it need not;  some packets (especially [control packets](https://en.wikipedia.org/wiki/802.11_Frame_Types)) are only 20 MHz wide, and some are 40.

After you recompile and run, run `nexutil -k` to see if the correct channel is set. And as I always advice, please skim through all the issues on this repo and the paper; it's less work than you think and saves a **lot** of time later. :)