seemoo-lab/nexmon_csi #150: capture los

tweigel-dev, 2020-11-24:
Hay, 
I use the nexmon_csi tool with the Raspbarry pi 3b plus at monitor mode and is confgured to gather csi filtered by the mac of the laptop.
I send a ping from a laptop to my router. 
The Raspbarry pi isn't connected to the wifi but captures csi values but i dont gather csi per response. 
If i ping 5 times per sec ( -i 0.2) it should send ~300 requests per minute. but i gather only ~100 csi values within this minute. 
My tests exterminate that the used channel or signal frequency has no impact to this behavior. 

i count them with tcpdump
```
sudo tcpdump -i wlan0 dst port 5500

and get this after one minute:

72 packets captured
105 packets received by filter
28 packets dropped by kernel

```
The TCPDUMP behaves like the gathering stops for a few seconds and starts again.

Why the kernel block some of them and where is the rest of the ~200 csi packages.
Did anyone has an idea why there is a CSI leakage?

zeroby0, 2020-11-24:
I've been facing this too! The CSI collected doesn't correspond with the number of packets going in the air.. with some routers.

If I'm collecting data when my laptop is connected to [Archer C80](https://www.tp-link.com/in/home-networking/wifi-router/archer-c80/), the Pi collects very little packets, and most of them are non-data types. With a filter to collect only data frames, I still only collect 4-5 packets a minute, and they are not ICMP.

With a [Archer C20](https://www.tp-link.com/in/home-networking/wifi-router/archer-c20/), the data collection corresponds much more closely to ICMP packets, especially with data-frame filters applied.

I have also observed that placing the Raspberry Pi a few inches above the ground results in collection of significantly more packets than when it's on the floor.

tweigel-dev, 2020-11-27:
Interesting. I read the free-wifi paper of this tool and notice the following table : 
[image]
there is a much higher output as i gather with my setup. 
I will try to figure out what could be the difference. Therefore I will test my environment with the Asus-router  instead of the pi3+.