seemoo-lab/nexmon_csi #151: Why use UDP datagram

junqing-zhang, 2020-11-26:
Hello, 

Firstly thank you for the wonderful tool. I would like to enquire an issue regarding the UDP datagram. As far as I know, UDP is higher layer while WiFi defines the physical and MAC layers. Therefore, a UDP datagram is not necessarily carried by WiFi transmissions. Is there any special consideration doing so?

Another related question is what WiFi frames can be captured. The WiFI MAC layer protocol defines different frames such as control, management and data frames. Are we able to capture the control and management frames? e.g., the ACK, probe request frames?

Thanks.

zeroby0, 2020-11-26:
Heyo!

> Another related question is what WiFi frames can be captured. 

**All** kinds frames are captured. No distinction is made unless you set a filter with `-b`. That doesn't mean every single frame in the air is captured, though. The radio is "deafened" when reading CSI so that a new frame doesn't overwrite the current value while reading it. Which _may_ lead to some kind of frames being captured more in some situations, but the firmware itself doesn't make any distinction.

> Why UDP frames

The UDP datagram is not related to the WiFi transmission. The correct answer to this question is to read the [paper](https://dl.acm.org/doi/pdf/10.1145/3349623.3355477?casa_token=4pZMCAs1FNkAAAAA:eLX3uSveBmwmwzbX0hJL1PEunGFlT5FekxxjuuHO96oQlWjyv02tSF-6eEUumGWuqLuhc0-GmZ8l), but I'll try to explain.

There are two chips in the broadcomm WiFi cards. A d11 core, and an ARM microprocessor. The d11 core interfaces with the radio, and decodes the frame. This frame is sent to the ARM processor which strips out the WLAN header and attaches an Ethernet header. There are other things the ARM processor does, but they are not of our concern. This packet with Ethernet header is sent to the Linux Kernel, which processes it as a normal Ethernet packet. The reverse is executed when transmitting a packet.

Nexmon_CSI modifies the WiFi firmware so that we get CSI measurements instead of the actual decoded packet. The software that is run in the d11 core is changed with the `.patch` files in the `src` folder. The d11 core now sends the CSI values instead of the actual packet. And the ARM processor runs `src/csi_extractor.c`. The ARM processor receives the CSI values from the d11 core, adds a UDP header to it, and puts it in an Ethernet packet. This new Ethernet packet is now sent to the kernel. We _could_ use TCP, but it's unnecessarily complex for our use case.

Which is why you see CSI data arriving inside UDP packets. They are just CSI data of all kinds of frames, embedded inside UDP frames so that we can easily read them using userland software like tcpdump. I simplified it a little bit, but more accurate and detailed working is available in the paper.

If you're planning to do any serious work with this software, I strongly suggest you read the paper and the issues created in this repo.

junqing-zhang, 2020-11-26:
Thanks for the detailed response. I will read the paper carefully. 
We would like to collect the CSI for security and wireless sensing research. As the Linux 802.11n CSI tool and Atheros CSI tool require the PCI interface, nexmon CSI extractor is preferred.

mzakharo, 2020-11-26:
> **All** kinds frames are captured. No distinction is made unless you set a filter with `-b`

I was unable to capture CSI on  'ACK' frames, at all, even though I could see them through the 'mon0' interface. I am pretty sure not 'all' kinds of 802.11 frames are captured, but only certain kinds are supported by the firmware.

zeroby0, 2020-11-26:
@mzakharo Good to see you again :)

You're right. I think some kinds of frames use modulation that's not compatible with csi extraction, too. I have seen csi on block_ack, but I've never seen ACK frames as well. I think I jumped on that statement in hindsight 😅

I see two `mov 0 DUMP_CSI` in bcm43455c0 patch. Is there anywhere else the flag could be set from?

mzakharo, 2020-11-26:
If ACK is sent via OFDM (which it is in 5Ghz), it can be used to retreive CSI (as confirmed by other OFDM-CSI extraction platforms). ACK's simply do not genreate CSI via nexmon patches, for reason yet unknown.  It would be nice to see support for ACK's since, well, these types of frames are the easiest to generate via a NULL-Data packet request from a Transmitter, and simplest to control data-rate with.

junqing-zhang, 2020-11-27:
Hello @zeroby0 , I have read your explanation and the paper carefully. I would like to check if I have the correct understanding. Here UDP datagram is used for data communication/transfer between the ARM FullMAC core and the Linux host. It is not a UDP datagram transmitting from one device to another one. Right?

zeroby0, 2020-11-27:
That's correct. @junqing-zhang