seemoo-lab/nexmon_csi #152: explicit or implicit beamforming

tweigel-dev, 2020-11-27:
Hay @matthiasseemoo @jlinktu and rest of the great community,
at research for beamforming in combination of CSI gathering, I want to know, which type of beamforming mechanism is used to implement nexmon CSI? 
dose it use Explicit or Implicit beamforming.
Snipped out of 802.11 ac:

> **Implicit  feedback**:  When  using  implicit  feedback,  the  beamformer  receives  long  training  symbolstransmitted by the HT beamformee, which allow the MIMO channel between the HT beamformeeand  HT  beamformer  to  be  estimated.  If  the  channel  is  reciprocal,  the  HT  beamformer  can  use  the training symbols that it receives from the HT beamformee to make a channel estimate suitable forcomputing the transmit steering matrix. Generally, calibrated radios in MIMO systems can improvereciprocity.
>**Explicit feedback**: When using explicit feedback, the HT beamformee makes a direct estimate of thechannel  from  training  symbols  sent  to  the  HT  beamformee  by  the  HT  beamformer.  The  HTbeamformee  may  prepare  CSI  or  steering  feedback  based  on  an  observation  of  these  trainingsymbols. The HT beamformee quantizes the feedback and sends it to the HT beamformer. The HTbeamformer can use the feedback as the basis for determining transmit steering vectors. See 9.29.3.

This is impotent because i want to gather SNR or SINR belong CSI and RSSI values(like Intel5300 CSI) and for this it is easier to use or extract with explicit beamforming. Because the explicit beamforming includes a feedback frame which contains the gathered csi and the messuared SNR/SINR.
snapshot out of 802.11 ac: (8.4.1.49 MU Exclusive Beamforming Report field)

[image]

 If I achieve this, I will try to contribute my results here.

jlinktu, 2020-11-27:
This tool extracts the channel estimate that is computed from the long training sequence which is part of an 802.11 OFDM frame's PHY preamble (see IEEE Std 802.11-2016 17.3.3, 19.3.9.3.4, 19.3.9.4.6, 21.3.8.2.3, 21.3.8.3.5), and makes no use of beamforming feedbacks.

tweigel-dev, 2020-11-30:
i may misunderstanding the 802.11 versions but the mechanism of training  had implemented at 802.11ah 2016. but nexmon works with 802.ac chips spezified 2013. how does this correlate? because at the standard 802.11 ac is no specification of such a training preamble.
[802.11ac](https://ieeexplore.ieee.org/document/7797535#G12.116856)

tweigel-dev, 2020-12-01:
@jlinktu  Are the CSI is gathered per frame and per Antanna or only per frame? because to calculate such things like AoA  it is necessary to use CSI data per antenna. The examples pcap at matlab or the image at the readMe didn't show CSI data per antenna.

jlinktu, 2020-12-01:
The 802.11 document you referenced is an amendment to the revision [IEEE Std 802.11-2007](https://ieeexplore.ieee.org/document/4248378), which already describes the OFDM preamble training fields in 17.3.3. But you might want to have a look at the [IEEE Std 802.11-2016](https://ieeexplore.ieee.org/document/7786995) revision instead as it also contains preamble description for HT in 19.3.9.3.4, 19.3.9.4.6 and VHT in 21.3.8.2.3, 21.3.8.3.5.

You can extract CSI on a per frame basis for each antenna and each spatial stream. The README uses the term "core" instead of "antenna".

tweigel-dev, 2020-12-01:
Thanks for your resonse @jlinktu !
I reread your paper of ["Free Your CSI: A Channel State Information Extraction Platform For Modern Wi-Fi Chipsets"](https://dl.acm.org/doi/10.1145/3349623.3355477) and I noticed that nexmon_csi use the fact that a few parts of the hardware at broadcom chips has direct access to the registers that stores the csi until the next frame is arriving (FullMAC ARM CPU and D11 MAC core). 
and Nexmon_csi implements a hook with the help of the OpenFWWF project, that reads each csi entry out of the registry if the mac-adress matchs. 
Isn't it?
That means that it has no direct touch with the beamforming feedback mechanism, which is used at intel-CSI and atheros-CSI to gather the CSI-Data. 
Thus, the SNR, Noise and other nice metrics, to evaluate wifi, are not reachable with this mechanism, isn't it?
Or did you think that this metrics are temporary stored in a register belong the CSI's with the same accassibility by FullMAC ARM CPU or D11 MAC core?