seemoo-lab/nexmon_csi #154: Unable to compile the firmware patch and install it on your RT-AC86U router.

ShivamJoshi64, 2020-11-30:
Hi,

I've been trying to follow the steps mentioned in the [getting started](https://github.com/seemoo-lab/nexmon_csi#bcm4366c0) section for the RT-AC86U router. However, when running the make install-firmware command I am getting an error saying connection refused. (please refer: Screenshot_1)

I am able to access the internet through this router but can not SSH into it. I guess I need to change some settings to enable SSH for this router but I am not able to key into the [router setting page(Web GUI)](https://www.asus.com/us/support/FAQ/1005263) after I have done the initial setup for the router. (please refer: Screenshot_2)

If someone has faced this issue before or knows how to get around it, please help. I'd be really grateful for your support, any suggestion is appreciated.

[image]
[image]

zeroby0, 2020-11-30:
Enter the IP of the router instead of `router.asus.com`.  I have the ac86 connected to another router via ethernet, and my laptop connected to the other router via wifi. So now I can see the ac86's IP in the other router, and can access it.

`router.asus.com` has no dns records. If you are connected to an asus router, the router intercepts your connection and serves it's configuration page.

ShivamJoshi64, 2020-12-09:
Hi @zeroby0 
Thank you so much for your help. I followed these steps and was able to access the AC86 router.

However, When I followed the steps mentioned in the [Usage](https://github.com/seemoo-lab/nexmon_csi/blob/master/README.md#usage) section, I am getting the following error:

> __nex_driver_io: error ret=-1 errno=95

Can you please assist me here? please refer the following screenshot:

[image]

zeroby0, 2020-12-09:
I think you should use eth6.

ShivamJoshi64, 2020-12-09:
It gives the same error.

zeroby0, 2021-05-08:
> __nex_driver_io: error ret=-1 errno=95

You should reload the kernel module and bring up the interfaces every time you reboot the router. You can also automate that: https://github.com/nexmonster/nexmon_csi/discussions/2