seemoo-lab/nexmon_csi #155: Help: sometimes fail to capture packets on Raspberry 4B

Colin-Guolin, 2020-12-01:
Hi Guys

I am using Raspberry pi 4B to capture CSI, but it is not stable, it works just fine some times, but it also fail to capture the packets from time to time and got zero packet captured. To make sure there are traffic on the channel, I used another Raspberry to connect to my router and ping to the router, and the receiver Raspberry monitor the same channel and use `-m MAC_ADDRESS` to make sure the packet comes from the transmitter.

But it can't always success, sometimes it stop to capture any packets after it captured some packets, sometimes it not capture any packets or not capture any packets after I closed the capture process by `ctrl + c` and start it again by 'tcpdump -i wlan0 dst port 5500 -vv -w CSI.pcap -c 1000'

here is my script to start up the capture:
`#!/bin/bash
echo " Start"

makecsiparams -c 36/80 -C 1 -N 1 -m dc:a6:32:49:56:15
pkill wpa_supplicant

echo "Killed Wpa_supplicant"
ifconfig wlan0 up

echo "wlan0 up"
nexutil -Iwlan0 -s500 -b -l34 -vKuABEQAAAQDcpjJJVhUAAAAAAAAAAAAAAAAAAAAAAAAAAA==

echo "nextil"
iw phy `iw dev wlan0 info | gawk '/wiphy/ {printf "phy" $2}'` interface add mon0 typ$

echo "iw phy"
ifconfig mon0 up

echo "mon0 up and collecting CSI"
tcpdump -i wlan0 dst port 5500 -vv -w CSI.pcap -c 1000`

What is the reason for this issue? Any advise to fix this problem? Thanks in advance

tweigel-dev, 2020-12-02:
I think it correlates with https://github.com/seemoo-lab/nexmon_csi/issues/150 ;-)

Brian0212, 2021-06-15:
Hi
I meet the same problem too.
Sometimes I can get CSI but after few try, I can't get CSI any more.
Is  there any advise for this problem?

zeroby0, 2021-06-15:
Try applying the amsdu fixes from pull 46, and reload the firmware into kernel after every reboot.

I have a fork here that does both of them: https://github.com/nexmonster/nexmon_csi/tree/pi-5.4.51

Brian0212, 2021-06-15:
@zeroby0  thx for your fast reply, and yes I have reloaded every time I reboot.
I have seem this pull. So is this problem  cause by null pointer?

zeroby0, 2021-06-15:
I'd think so. It fits the symptoms. :)

Brian0212, 2021-06-15:
@zeroby0 thanks a lot.
I will try your fork later.
Appreciate

Brian0212, 2021-06-16:
@zeroby0 thx for your help, after using your fork, I can get CSI.
But I have a question, how do you parse the pcap data.
I have read "https://github.com/Gi-z/CSIKit" that said the data format of yours is different from original one.
So should I using CSIKit, the original matlab code or  csiread ?
Appreciate

zeroby0, 2021-06-16:
@Brian0212 

So the branch I linked you to uses the original data format, and should be parse-able by CSIkit, and by the matlab code. I have also written a parser optimised for speed, [it's available here](https://github.com/nexmonster/nexmon_csi/tree/feature/python/utils/python).

There is an other branch called [5.4.51-**plus**](https://github.com/nexmonster/nexmon_csi/tree/pi-5.4.51-plus/#analyzing-the-csi) which uses a different data format with more metrics like RSSI. To parse the data from that branch, you should use [csi-explorer](https://github.com/zeroby0/csi-explorer).

Brian0212, 2021-06-16:
Ohhh I see.
Sorry I don't notice the branch is different.

Thanks for your help.