seemoo-lab/nexmon_csi #157: Bug: Core and Spatial Stream is always 0x0000 in PCAP files

zeroby0, 2020-12-03:
**Type**: bug
**Priority**: low

### Description
The Core and Spatial Stream bytes are always `0x0000` with the latest commit on master. The [expected value](https://github.com/seemoo-lab/nexmon_csi#analyzing-the-csi) is `0b00001001` or `0x0009` as makecsiparams is set with `-C 1 -N 1`.

Here is a screen shot from Wireshark.
[image]

Here is my zip compressed pcap file: [master-80.zip](https://github.com/seemoo-lab/nexmon_csi/files/5635953/master-80.zip)

### Configuration
**Device**: Raspberry Pi 4B
**Operating System**: [Buster 2020-08-20](https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2020-08-24/) with Kernel 5.4.51.
**Channel**: 36. Not tested on other channels.
**Bandwidth**: 80 MHz and 40 MHz. Not tested on 20 MHz.

This is a low priority bug because Raspberry Pi has only one antenna and one wifi core, so these bytes are not really applicable on Raspberry Pi devices. However, the behavior deviates from the documentation, and is likely to cause confusion among beginners.

I have not verified if the values are hardcoded to `0x0000` or if this is the result of an underlying bug yet. If it's hardcoded, can we correct it in the next commit?

sie2, 2020-12-20:
Are you using two devices? I also noticed that the sequence number seems very erratic.

zeroby0, 2020-12-20:
> Are you using two devices? I also noticed that the sequence number seems very erratic.

No, I'm using one device. This bug is about Core and Spatial Stream bytes, but the Sequence Number bytes may seem erratic because it contains both Sequence and Fragment number. Most packets have only 1 fragment, so the number shifts by 16.

If you need to separate Sequence and Fragment numbers, cast the bytes to unsigned integer, and then:
```Python
sc = int.from_bytes(#uint16: SC
    <Sequence Number Bytes>,
    byteorder = 'little',
    signed = False
)
fn = sc % 16 # Fragment Number
sc = int((sc - fn)/16) # Sequence Number
```
https://github.com/zeroby0/nexmon_csi/blob/37665448f975aa4ed1883b73ef1debe3aa826699/utils/python/decoders/interleaved.py#L93

But even accounting for that, Sequence Number could be erratic because the radio is "deafened" when reading CSI of a packet. So packets that arrive while the CSI value is read are missed.

sie2, 2020-12-20:
> No, I'm using one device.

That's weird. I seem to be reading addresses `3c:84:6a:d0:d6:b2` and `40:ec:99:cc:d9:8b` from that .pcap file.

>  Sequence Number bytes may seem erratic because it contains both Sequence and Fragment number

I learned something new today.

>  This bug is about Core and Spatial Stream bytes

I just tested this on my Raspberry Pi and got similar results.  

The other weird thing is, I tested in on the RT-AC86U and it does report the core and stream, but on the high bytes instead of the low bytes? From the README.md, it says

> The next two bytes contain core and spatial stream number where the lowest three bits indicate the core and the next three bits the spatial stream number, e.g. 0x0019 (0b00011001) means core 0 and spatial stream 3.

But when I set `-C -0xf` and `-N 0xf` in `makecsiparams`, I get values like `0b0001101100000000` in core+stream component of the packets. Is this your experience as well?