seemoo-lab/nexmon_csi #159: unknown CSI waveforms after a clean channel, single spatial stream, single frame type filter

yujianyuanhaha, 2020-12-06:
greetings,

I recently apply a CSI collection with filter like `./makecsiparams -c 8/20 -C 0x1 -N 0x1 -m  04:D9:F5:75:AF:10 -b 0x88`, but the CSI is pretty noisy for unknown reason.

Plot all packages waveform together 
[image]

While plot them over time, e.g. plot 1-10th 100-110th ... packages's waveform
[image]

We can at least see 3 types of waveform, which confuse me a bit since I have applied all filter I can do ...

yujianyuanhaha, 2020-12-21:
it looks like the unknown frame come from noise, it is also discussed in paper [CRISLoc: Reconstructable CSI Fingerprintingfor Indoor Smartphone Localization](https://arxiv.org/abs/1910.06895).

[image]

yael1994, 2021-03-03:
Can you explain how you create the graphs? I record csi data and I want to present them like you did.

Gi-z, 2021-03-03:
> Can you explain how you create the graphs? I record csi data and I want to present them like you did.  

You can view subcarrier curves like these with [csi-explorer](https://github.com/zeroby0/csi-explorer).

zeroby0, 2021-03-04:
csi-explorer  needs a different format of pcap, and is a little slow. I made a faster version that works with the master branch of Nexmon_csi: https://github.com/zeroby0/nexmon_csi/tree/feature/python/utils/python.

There are still a few improvements it needs, after which we will make a pull request. I have been much busy for the past few weeks, so the progress is slow, but it should be ready in a week or two.