seemoo-lab/nexmon_csi #160: indistinguishable CSI waveform on NEXUS 5

yujianyuanhaha, 2020-12-07:
greetings,

for those collecting CSI on NEXUS5, have anyone meet the problem that the **CSI of the four spatial stream** (assume the TX is ASUS router with 4 antennas, and RX/ Sniffer is NEXUS 5 so there are 4x1 spatial stream) **is pretty the same?**    

I assume they should be different since my CSI collection on ASUS router, all 4x4=16 spatial stream's CSI is different from each other.

I even install the extensible antennas to make the space between each TX antennas larger, but the waveform still pretty close.

Anyone kindly could look into the problem a bit?

Thanks in advance.

yangtat, 2020-12-09:
On the Nexus 5, the CSI of the four spatial streams I collected are almost identical too. But  on the Nexus 6P I can get different ones.

yujianyuanhaha, 2020-12-09:
@yangtat thanks, that might be a solution
however, the Tx-by-RX (here is 4 x 1 ) spatial stream, I have separate the antenna as I can ...