seemoo-lab/nexmon_csi #161: Cannot insmod dhd.ko for ASUS RT-AC68U

zukky619, 2020-12-07:
Hi, I'm trying to compile nexmon_csi tool for ASUS RT-AC68U.
But, I met an error saying that "insmod: can't insert '/jffs/dhd.ko': invalid module format" at STEP 7. It seems that I did not make kernel module properly on Host PC. I hope someone can help me.

Host PC OS: Ubuntu 16.04LTS[4.4.0-21-generic] and Ubuntu 18.04LTS[5.3.0-28-generic] (I tried both and got same result.)
RT-AC68U kernel: 2.6.36.4bcmarm
RT-AC68U firmware version: 3.0.0.4.385_20633-g593a8ef

```
root@mmw-desktop:~/tmp/nexmon/patches/bcm4366c0/10_10_122_20/nexmon_csi# make install-firmware REMOTEADDR=192.168.24.1

          ###########   ###########   ##########    ##########           
         ############  ############  ############  ############          
         ##            ##            ##   ##   ##  ##        ##          
         ##            ##            ##   ##   ##  ##        ##          
         ###########   ####  ######  ##   ##   ##  ##    ######          
          ###########  ####  #       ##   ##   ##  ##    #    #          
                   ##  ##    ######  ##   ##   ##  ##    #    #          
                   ##  ##    #       ##   ##   ##  ##    #    #          
         ############  ##### ######  ##   ##   ##  ##### ######          
         ###########    ###########  ##   ##   ##   ##########           

            S E C U R E   M O B I L E   N E T W O R K I N G               

                               presents:                                  

              # ###   ###  #   # # ###  ###   ###  # ###                  
              ##   # #   #  # #  ##   ##   # #   # ##   #                 
              #    # #####   #   #    #    # #   # #    #                 
              #    # #      # #  #    #    # #   # #    #                 
              #    #  #### #   # #    #    #  ###  #    #                 

                The C-based Firmware Patching Framework                   

                           !!! WARNING !!!                                
    Our software may damage your hardware and may void your hardware’s    
     warranty! You use our tools at your own risk and responsibility      

  COLLECTING STATISTICS read /home/mmw/tmp/nexmon/STATISTICS.md for more information
  COMPILING src/ioctl.c => obj/ioctl.o (details: log/compiler.log)
  COMPILING src/csi_extractor.c => obj/csi_extractor.o (details: log/compiler.log)
  COMPILING src/patch.c => obj/patch.o (details: log/compiler.log)
  COMPILING src/regulations.c => obj/regulations.o (details: log/compiler.log)
  COMPILING src/console.c => obj/console.o (details: log/compiler.log)
  COMPILING src/version.c => obj/version.o (details: log/compiler.log)
  DISASSEMBLING UCODE /home/mmw/tmp/nexmon/firmwares/bcm4366c0/10_10_122_20/nonmuucode.bin => gen/ucode.asm
  PATCHING UCODE src/csi.ucode.bcm4366c0.10_10_122_20.patch => src/csi.ucode.bcm4366c0.10_10_122_20.asm
  ASSEMBLING UCODE src/csi.ucode.bcm4366c0.10_10_122_20.asm => gen/ucode.bin
  COPYING NON MU UCODE gen/ucode.bin => gen/nonmuucode.bin
  COMPRESSING UCODE gen/nonmuucode.bin => gen/nonmuucode_compressed.bin
  GENERATING C FILE gen/nonmuucode_compressed.bin => src/nonmuucode_compressed.c
  COMPILING src/nonmuucode_compressed.c => obj/nonmuucode_compressed.o (details: log/compiler.log)
  COPYING NON MU UCODE X /home/mmw/tmp/nexmon/firmwares/bcm4366c0/10_10_122_20/nonmuucodex.bin => gen/nonmuucodex.bin
  COMPRESSING UCODE gen/nonmuucodex.bin => gen/nonmuucodex_compressed.bin
  GENERATING C FILE gen/nonmuucodex_compressed.bin => src/nonmuucodex_compressed.c
  COMPILING src/nonmuucodex_compressed.c => obj/nonmuucodex_compressed.o (details: log/compiler.log)
  COPYING MU UCODE /home/mmw/tmp/nexmon/firmwares/bcm4366c0/10_10_122_20/muucode.bin => gen/muucode.bin
  COMPRESSING UCODE gen/muucode.bin => gen/muucode_compressed.bin
  GENERATING C FILE gen/muucode_compressed.bin => src/muucode_compressed.c
  COMPILING src/muucode_compressed.c => obj/muucode_compressed.o (details: log/compiler.log)
  COPYING MU UCODE X /home/mmw/tmp/nexmon/firmwares/bcm4366c0/10_10_122_20/muucodex.bin => gen/muucodex.bin
  COMPRESSING UCODE gen/muucodex.bin => gen/muucodex_compressed.bin
  GENERATING C FILE gen/muucodex_compressed.bin => src/muucodex_compressed.c
  COMPILING src/muucodex_compressed.c => obj/muucodex_compressed.o (details: log/compiler.log)
  COMPILING /home/mmw/tmp/nexmon/patches/common/ucode_compression_code.c => obj/ucode_compression_code.o (details: log/compiler.log)
  COMPILING /home/mmw/tmp/nexmon/patches/common/wrapper.c => obj/wrapper.o (details: log/compiler.log)
  COMPILING /home/mmw/tmp/nexmon/patches/common/argprintf.c => obj/argprintf.o (details: log/compiler.log)
  COMPILING /home/mmw/tmp/nexmon/patches/common/helper.c => obj/helper.o (details: log/compiler.log)
  COMPILING /home/mmw/tmp/nexmon/patches/common/udptunnel.c => obj/udptunnel.o (details: log/compiler.log)
  COMPILING /home/mmw/tmp/nexmon/patches/common/objmem.c => obj/objmem.o (details: log/compiler.log)
  COMPILING /home/mmw/tmp/nexmon/patches/common/securitycookie.c => obj/securitycookie.o (details: log/compiler.log)
  COMPILING /home/mmw/tmp/nexmon/patches/common/radiotap.c => obj/radiotap.o (details: log/compiler.log)
  COMPILING /home/mmw/tmp/nexmon/firmwares/bcm4366c0/10_10_122_20/flashpatches.c => obj/flashpatches.o (details: log/compiler.log)
  PREPARING gen/nexmon.pre => gen/nexmon2.pre
  GENERATING LINKER FILE gen/nexmon.pre => gen/nexmon.ld
  GENERATING LINKER FILE gen/nexmon.pre => gen/flashpatches.ld
  GENERATING LINKER FILE gen/memory.ld
  LINKING OBJECTS => gen/patch.elf (details: log/linker.log, log/linker.err)
  GENERATING MAKE FILE gen/nexmon.pre => gen/nexmon.mk
  GENERATING MAKE FILE gen/nexmon.pre => gen/flashpatches.mk
  APPLYING FLASHPATCHES gen/flashpatches.mk => dlarray_4366c0.bin (details: log/flashpatches.log)
  APPLYING PATCHES gen/nexmon.mk => dlarray_4366c0.bin (details: log/patches.log)
  COPYING TO ROUTER dhd.ko => /jffs/dhd.ko
admin@192.168.24.1's password: 
dhd.ko                                                          100% 1684KB 842.1KB/s   00:02    
  LOADING /jffs/dhd.ko
admin@192.168.24.1's password: 
rmmod: can't unload module 'dhd': No such file or directory
insmod: can't insert '/jffs/dhd.ko': invalid module format
Makefile:356: recipe for target 'install-firmware' failed
make: *** [install-firmware] Error 8
```

zeroby0, 2020-12-07:
More curiously, rmmod failed:
`rmmod: can't unload module 'dhd': No such file or directory`. You're sure `192.168.24.1` is your router, right?

I'll post my dhd.ko in a while. Try insmod-ing that file to see if it works.

zukky619, 2020-12-07:
I recognize a fatal difference...It's not RT-AC86U but RT-AC68U...
I order RT-AC86U but store made a mistake.
I'm sorry for upsetting you.

closed 2020-12-07 by zukky619