seemoo-lab/nexmon_csi #162: Cannot insmod dhd.ko for ASUS RT-AC88U

YogiWu, 2020-12-07:
Hi, I'm trying to compile nexmon_csi tool for ASUS RT-AC88U.
But, I met an error that "insmod: can't insert '/jffs/dhd.ko': invalid module format" at STEP 7. The ASUS RT-AC88U use the bcm4366e chip as the same ASUS RT-AC86U.I hope someone can help me.

`          ###########   ###########   ##########    ##########           
         ############  ############  ############  ############          
         ##            ##            ##   ##   ##  ##        ##          
         ##            ##            ##   ##   ##  ##        ##          
         ###########   ####  ######  ##   ##   ##  ##    ######          
          ###########  ####  #       ##   ##   ##  ##    #    #          
                   ##  ##    ######  ##   ##   ##  ##    #    #          
                   ##  ##    #       ##   ##   ##  ##    #    #          
         ############  ##### ######  ##   ##   ##  ##### ######          
         ###########    ###########  ##   ##   ##   ##########           

            S E C U R E   M O B I L E   N E T W O R K I N G               

                               presents:                                  

              # ###   ###  #   # # ###  ###   ###  # ###                  
              ##   # #   #  # #  ##   ##   # #   # ##   #                 
              #    # #####   #   #    #    # #   # #    #                 
              #    # #      # #  #    #    # #   # #    #                 
              #    #  #### #   # #    #    #  ###  #    #                 

                The C-based Firmware Patching Framework                   

                           !!! WARNING !!!                                
    Our software may damage your hardware and may void your hardware’s    
     warranty! You use our tools at your own risk and responsibility      

  COLLECTING STATISTICS read /home/yogi/Project/2020/router/nexmon/STATISTICS.md for more information
  COMPILING src/version.c => obj/version.o (details: log/compiler.log)
  PREPARING gen/nexmon.pre => gen/nexmon2.pre
  GENERATING LINKER FILE gen/nexmon.pre => gen/nexmon.ld
  GENERATING LINKER FILE gen/nexmon.pre => gen/flashpatches.ld
  LINKING OBJECTS => gen/patch.elf (details: log/linker.log, log/linker.err)
  GENERATING MAKE FILE gen/nexmon.pre => gen/nexmon.mk
  GENERATING MAKE FILE gen/nexmon.pre => gen/flashpatches.mk
  APPLYING FLASHPATCHES gen/flashpatches.mk => dlarray_4366c0.bin (details: log/flashpatches.log)
  APPLYING PATCHES gen/nexmon.mk => dlarray_4366c0.bin (details: log/patches.log)
  COPYING TO ROUTER dhd.ko => /jffs/dhd.ko
admin@192.168.50.1's password: 
dhd.ko                                                                    100% 1684KB   1.4MB/s   00:01    
  LOADING /jffs/dhd.ko
admin@192.168.50.1's password: 
insmod: can't insert '/jffs/dhd.ko': invalid module format
Makefile:356: recipe for target 'install-firmware' failed
make: *** [install-firmware] Error 8
`

tweigel-dev, 2020-12-10:
can you please execute the following command and share the output with us ?

```
file ./dhd.ko
```

tweigel-dev, 2020-12-10:
i found the reason. you are trying to use an 802.11 ax device. 
http://en.techinfodepot.shoutwiki.com/wiki/ASUS_RT-AX88U
look into the table of supported chips. the BCM43684 isn't supported yet.

zeroby0, 2020-12-10:
@tweigel-dev 
But they said they're using ASUS RT-AC88U?
> The ASUS RT-AC88U use the bcm4366e chip as the same ASUS RT-AC86U.

The RT-AC86U uses bcm4366c0, which is an older revision than 4366e. Maybe the dhd modules are not interchangeable between devices. That said, [@maxedbeech got GT-AC5300 working](https://github.com/seemoo-lab/nexmon_csi/issues/80), which uses the bcm4366e chip (https://deviwiki.com/wiki/ASUS_GT-AC5300), so it should work.

Can you try to insmod the dhd.ko from here: https://github.com/nexmonster/nexmon_csi/discussions/2? It's tested to work, and may work if your dhd.ko was somehow not built properly (For example, the compiler was not set properly).

tweigel-dev, 2020-12-10:
@zeroby0 I think the ASUS RT-AC88U did not have the bcm4366 because it is an AX Router. and the bcm4366 is a AC chip. 

> The ASUS RT-AC88U use the bcm4366e chip as the same ASUS RT-AC86U.
This seems to be a wrong information.

zeroby0, 2020-12-10:
The ASUS RT-AC88U is an AC router: https://www.asus.com/in/Networking/RT-AC88U/specifications/

The RT-AX88U, however, is an AX router with a different chip.

YogiWu, 2020-12-11:
@zeroby0 Thanks for your help.I followed the step in [nexmonster#2](https://github.com/nexmonster/nexmon_csi/discussions/2) but it still told me that `insmod: can't insert '/jffs/dhd.ko': invalid module format`.I even tried to compile the dhd.ko in raspberry pi but alse got the same result.It's sure that RT-AC88U is not supported now.

> @tweigel-dev
> But they said they're using ASUS RT-AC88U?
> 
> > The ASUS RT-AC88U use the bcm4366e chip as the same ASUS RT-AC86U.
> 
> The RT-AC86U uses bcm4366c0, which is an older revision than 4366e. Maybe the dhd modules are not interchangeable between devices. That said, [@maxedbeech got GT-AC5300 working](https://github.com/seemoo-lab/nexmon_csi/issues/80), which uses the bcm4366e chip (https://deviwiki.com/wiki/ASUS_GT-AC5300), so it should work.
> 
> Can you try to insmod the dhd.ko from here: [nexmonster#2](https://github.com/nexmonster/nexmon_csi/discussions/2)? It's tested to work, and may work if your dhd.ko was somehow not built properly (For example, the compiler was not set properly).

tweigel-dev, 2020-12-11:
@YogiWu 
can you copy a existing executable at  from your router to your laptop , like /bin/bash  with `scp` and run the `file` command. Maybe your router runs with an other CPU architecture. x86 or arm processor? or an 32 bit vs 64 bit  issue

YogiWu, 2020-12-13:
@tweigel-dev I execute some command and get this:
```
admin@RT-AC88U-A1B0:/tmp/home/root# uname -a
Linux RT-AC88U-A1B0 2.6.36.4brcmarm #1 SMP PREEMPT Tue Aug 11 08:49:06 CST 2020 armv7l ASUSWRT
admin@RT-AC88U-A1B0:/tmp/home/root# cat /proc/cpuinfo 
Processor	: ARMv7 Processor rev 0 (v7l)
processor	: 0
BogoMIPS	: 2798.38

processor	: 1
BogoMIPS	: 2798.38

Features	: swp half thumb fastmult edsp 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x3
CPU part	: 0xc09
CPU revision	: 0

Hardware	: Northstar Prototype
Revision	: 0000
Serial		: 0000000000000000
admin@RT-AC88U-A1B0:/tmp/home/root# cat /proc/version 
Linux version 2.6.36.4brcmarm (root@asus) (gcc version 4.5.3 (Buildroot 2012.02) ) #1 SMP PREEMPT Tue Aug 11 08:49:06 CST 2020
yogi @ yogi-PC in ~:file dhd.ko
dhd.ko: ELF 64-bit LSB relocatable, ARM aarch64, version 1 (SYSV), BuildID[sha1]=8c25600e4b1ff34cb06b348b6f862f9796207c51, not stripped
```
The firmwork version of my ASUS AC88U is the latest one on the ASUS website.It seems that armv7l may be 32-bit which does not match the dhd.ko module.

zeroby0, 2020-12-13:
Hey @maxedbeech!

How did you get nexmon_csi working for GT-AC5300? Did you have to modify anything?
We're trying to get nexmon_csi working on another router with bcm4366e.

maxbeech, 2020-12-13:
Hey @zeroby0! 

### To answer your question...
With the fantastic guidance of @jlinktu, having compared two dhd.ko files, I got the GT-AC5300 working by using the patched dhd.ko originally meant for the ac86u. This is because we found that the dhd.ko of the GT-AC5300 contains firmware version 10.10.122.20 compiled in 2017 (the ac86u firmware is also 10.10.122.20 but compiled in 2019).

### My progress on trying to get the RT-AC88U to work...
Funnily enough I've been trying to get the RT-AC88U working and am facing the exact same error.
I again liaised with the generous @jlinktu on this and below is a summary of the progress we made:

**Jakob:**
> The wifi firmware itself (it is embedded inside the kernel module) should work as the RT-AC88U has the same wifi chip as the RT-AC86U.
> However, they have different main CPU architectures. The RT-AC86U has a BCM4906 which is a 64bit ARM Cortex-A53, whereas the RT-AC88U comes with a BCM4709C0 which is a 32bit ARM Cortex-A9.
> The dhd.ko provided in the repository is taken from an RT-AC86U and thus indeed incompatible with the RT-AC88U.
> However, if you are lucky, it might be possible to embed the same firmware inside a dhd.ko compatible with the RT-AC88U.
> On a RT-AC88U, running a default router firmware, search for the installed dhd.ko (e.g. with "find / -type f -name dhd.ko").
> Best would be to find a router firmware (if there is any) where the dhd.ko has the wifi firmware of version 10.10.122.20 embedded. (you can check the version with "strings dhd.ko | grep Version").
> If you find one you can send it to me so I can check if it is possible to embed the modified firmware inside the compatible kernel module.

**Me:**
Sent over the dhd.ko files corresponding to the following router versions:

- 10.10.122.47 (Firmware Version 3.0.0.4.384.21045)
- 10.10.122.303 (Firmware Version 3.0.0.4.385.20631)

**Jakob:**

> unfortunately the firmwares inside the two modules you sent me are smaller than the one of the RT-AC86U, making it not so easy to replace them.
> There is another thing you can try.
> Follow https://github.com/seemoo-lab/nexmon_csi#bcm4366c0 up to step 7. Instead of "make install-firmware ..." run "make dlarray_4366c0.bin".
> This should output a file "dlarray_4366c0.bin". Copy this file to your RT-AC88U (scp dlarray_4366c0.bin admin@<your routers address>:/jffs/).
> Try loading the original module with the modified firmware:
> ssh admin@<your routers address> "/sbin/rmmod dhd; /sbin/modprobe dhd firmware_path=/jffs/dlarray_4366c0.bin"
> If there's no error message you can verify the correct firmware was loaded by looking at the output of "dmesg" on the router.
> There should be a line containing "nexmon.org/csi".
> 
> However, I'm not sure if the module really handles the parameter.. but it is worth a try.

**Me:**
Attempted this and it reading the console read-out, it seemed to be successful. My assumption was then that I should run through the steps again, however I just got the same error on Step 7. 

It may be that I shouldn't have just gone through the steps again but instead have done something different, am not sure. So would recommend trying the same steps and seeing if you have more success than I did!

maxedbeech, 2020-12-13:
Hey @zeroby0! 

### To answer your question...
With the fantastic guidance of @jlinktu, having compared two dhd.ko files, I got the GT-AC5300 working by using the patched dhd.ko originally meant for the ac86u. This is because we found that the dhd.ko of the GT-AC5300 contains firmware version 10.10.122.20 compiled in 2017 (the ac86u firmware is also 10.10.122.20 but compiled in 2019).

### My progress on trying to get the RT-AC88U to work...
Funnily enough I've been trying to get the RT-AC88U working and am facing the exact same error.
I again liaised with the generous @jlinktu on this and below is a summary of the progress we made:

**Jakob:**
> The wifi firmware itself (it is embedded inside the kernel module) should work as the RT-AC88U has the same wifi chip as the RT-AC86U.
> However, they have different main CPU architectures. The [...]

**Me:**
Sent over the dhd.ko files corresponding to the following router versions:

- 10.10.122.47 (Firmware Version 3.0.0.4.384.21045)
- 10.10.122.303 (Firmware Version 3.0.0.4.385.20631)

**Jakob:**

> unfortunately the firmwares inside the two modules you sent me are smaller than the one of the RT-AC86U, making it not so easy to replace them.
> There is another thing you can try.
> Follow [...]

**Me:**
Attempted this and it reading the console read-out, it seemed to be successful. My assumption was then that I should run through the steps again, however I just got the same error on Step 7. 

It may be that I shouldn't have just gone through the steps again but instead have done something different, am not sure. So would recommend trying the same steps and seeing if you have more success than I did!

Additionally, I did only try Jakob's first idea with the two versions mentioned, so although unlikely to find one compatible with the RT-AC86U, it might be worth trying the other ones.

YogiWu, 2020-12-16:
Hey,I bougth the AC86U router and succesfully installed the firmwork on it.I have some question about how to connect the router after that.The router seems like a monitor.So should I use another wireless router as an signal source?

maxedbeech, 2020-12-16:
Hey @YogiWu! 

Huge congrats on getting the firmware going on it! How did you manage to do so?

Yes that’s right, you’ll need another router to broadcast a signal.

zeroby0, 2020-12-16:
@maxedbeech Thanks for the very nice explanation 😄 Somehow I didn't realise the processor in the router matters too, I thought only the WiFi chip mattered 😂 

> How did you manage to do so?

If you're trying to get started with rt-ac86u, I have precompiled binaries here: https://github.com/nexmonster/nexmon_csi/discussions/2. You can just copy them to the router instead of building nexmon from source. Makecsiparams is included too.

YogiWu, 2020-12-19:
I still confuse about some parts of README.What is the shaking the transmitter?How can I get the signal broadcast by other router?I conjecture that it will send the frame to other router and other router will reply a frame according to 801.11n protocol so that we can calculate the CSI.But according to the paper refered in the README,the patches get the data from the chip which is estimated by measuring the collected signal.Is that means it will collect all the frame of the signal with the specify frequency?Please someone tell me the details.

maxedbeech, 2020-12-19:
Thanks so much for your work on that @zeroby0 - legend! I won’t be able to try it out for about a month, but really looking forward to doing so!

maxedbeech, 2020-12-19:
> I still confuse about some parts of README.What is the shaking the transmitter?How can I get the signal broadcast by other router?I conjecture that it will send the frame to other router and other [...]

I’m definitely not the best person to answer this. I’d recommend reading more into csi and router frames and starting a new issue as it’s rather off-topic for this one. Good luck!

matthiasseemoo, 2020-12-26:
Hi Yogi,

i am not really shure what you want to know. CSI measures the effects on phase and amplitude introduced by the physical wireless channel between transmitter and receiver. Those effects need to be compensated when decoding the data transmitted in each physical wifi frame. So the wifi chip, by default, will always extract CSI and use it internally, we just found a way to extract it and pass it on to the user to make use of it.

Matthias

YogiWu, 2020-12-28:
Thanks for your answer.I already understood the fundamental of this project after reading some paper about wifi csi collection.