seemoo-lab/nexmon_csi #163: pi 4 can't collect CSI packets

parisafm, 2020-12-13:
I have installed nexmon CSI and I want to collect CSI with two raspberry Pi 4.
I use PC to send ping packets to AP (which is pi4). The AP is connected to PC and the other Pi4.
Although the PC sends ping packets to the AP, the second pi4 can't capture any CSI from listening to pong packets.
by the way nexmon is only installed on Pi4 client.
i appreciate your help in this matter.

zeroby0, 2020-12-13:
Did you reboot the pi after installing? You will need to reinstall the firmware after rebooting.

If that's not the case, post your forensics file (`cat /sys/kernel/debug/ieee80211/$(iw dev wlan0 info | gawk '/wiphy/ {printf "phy" $2}')/forensics`) and output of `dmesg`. They usually have indications of why the firmware stopped.

parisafm, 2020-12-14:
Yes, I rebooted pi
Here's the things u asked for :
cat /sys/kernel/debug/ieee80211/$(iw dev wlan0 info | gawk '/wiphy/ {printf "phy" $2}')/forensics

2 bytes to the heap
000000.148 wlc_bmac_info_init: host_enab 0
000000.148 Overriding vendor id = 0x14e4
000000.148 Overriding device id = 0x43ab
000000.149 wlc_bmac_attach, deviceid 0x43ab nbands 2
000000.171 wl0: wlc_bmac_attach: chiprev 6 corerev 54 cccap 0x18680009 maccap 0xa0018305 band 2.4G, phy_type 11 phy_rev 20
000000.172 wl0: wlc_bmac_attach: chiprev 6 corerev 54 cccap 0x18680009 maccap 0xa0018305 band 5G, phy_type 11 phy_rev 20
000000.198 wl0: wlc_stf_txcore_shmem_write: No clock
000000.199 wl0: wlc_ampdu_tx_set: AGG Mode = MAC+AQM txmaxpkts 0 txmaxpkts_agg 1024
000000.204 wl0: wlc_channels_commit: no valid channel for "#n" nbands 2 bandlocked 0
000000.205 wl0: Broadcom BCM4345 802.11 Wireless Controller 7.45.189 (r714228 CY)
000000.206 TCAM: 256 used: 237 exceed:0
000000.207 reclaim section 1: Returned 118076 bytes to the heap
000000.207 reclaim section 4: Returned 44 bytes to the heap
000000.207 sdpcmd_dpc: Enable
000000.231 wl0: wlc_iovar_op: txbf BCME -23 (Unsupported)
000000.233 wl0: unable to find iovar "rsdb_mode"
000000.233 wl0: wlc_iovar_op: rsdb_mode BCME -23 (Unsupported)
000000.234 wl0: unable to find iovar "tdls_enable"
000000.234 wl0: wlc_iovar_op: tdls_enable BCME -23 (Unsupported)
000387.095 wl0: unable to find iovar "toe_ol"
000387.095 wl0: wlc_iovar_op: toe_ol BCME -23 (Unsupported)
000387.095 wl0: wl_open
000387.106 wl0: wlc_phy_set_regtbl_on_femctrl: FIXME bt_coex
000387.123 wl0: unable to find iovar "nd_hostip_clear"
000387.123 wl0: wlc_iovar_op: nd_hostip_clear BCME -23 (Unsupported)
000454.826 wl0: wlc_phy_set_regtbl_on_femctrl: FIXME bt_coex
000478.744 wl0: wlc_phy_set_regtbl_on_femctrl: FIXME bt_coex
000563.667 wl0: wlc_phy_set_regtbl_on_femctrl: FIXME bt_coex
000833.525 wl0: wlc_phy_set_regtbl_on_femctrl: FIXME bt_coex
000843.132 wl0: wlc_phy_set_regtbl_on_femctrl: FIXME bt_coex
000882.717 wl0: wlc_phy_set_regtbl_on_femctrl: FIXME bt_coex
000894.624 wl0: unable to find iovar "toe_ol"
root@raspberrypi:/home/pi# p: toe_ol BCME -23 (Unsupported)

zeroby0, 2020-12-14:
If you reboot the printer, you will need to reload the firmware. Run these commands to make Pi automatically reload firmware on restart:

https://github.com/nexmonster/nexmon_csi/blob/e144fc47ff95b661fcc69b6ee9866f30e0d41cfd/install.sh#L62-L66

If that doesn't help either, try reinstalling nexmon with instructions as listed here: https://github.com/nexmonster/nexmon_csi/tree/pi-5.4.51

parisafm, 2020-12-26:
I tried to reinstall raspian as u said. Following all the instructions in https://github.com/zeroby0/nexmon_csi/tree/pi-4.19.97 ,unfortunately this error occurred:

root@raspberrypi:/home/pi# ifconfig wlan0 up
wlan0: ERROR while getting interface flags: No such device

tweigel-dev, 2020-12-26:
wlan0 is the alias for the WiFi nic . List the nics with iwconfig or ifconfig and Exchange wlan0 with yours

parisafm, 2020-12-28:
I reinstalled raspian from https://github.com/nexmonster/nexmon_csi/tree/pi-5.4.51 . Fortunately the wlan issue has been solved. 
As I mentioned above i try to use another Pi4 as Access Point.
I try to connect to AP with these two commands:

Sudo iwconfig wlan0 essid myAPname
Sudo dhclient wlan0

Everything seems fine and My second pi4 is connected to AP ( i use wireless network watcher to check connections)
But when i try to turn on MONITOR Mode, my second pi4 is disconnected.

tweigel-dev, 2020-12-28:
you cant be stay connected to a network at monitor mode. you have 2 possibilitys to gather csi:

#### 1
sent frames to the mac  nexmon raspi with the other device also at monitormode with tools like scapy.

#### 2
chose a setup where you "sniff" the frames of a communication of to other devices then your nexmon device and configure nexmon to filter by the mac of the receiving device.

------------------------

 @zeroby0  had done some trys to use a different wifi usb interface  at nexmon device but he failed, if if imagine correctly.

parisafm, 2020-12-28:
@tweigel-dev thanks for your response. 
i really appreciate your help
Could you please explain more about these two options? 
And which one do u recommend?

tweigel-dev, 2020-12-28:
It depends on your use-case? why you want to gather CSI values ?

parisafm, 2020-12-28:
I want to use CSI values for activity recognition. 
I do some activities(walk,fall down,...)  between two raspberries and  collect CSI values.

tweigel-dev, 2020-12-28:
Then both should work. 
If you have no expirience with monitormode and sniffing and faking layer2 frames use the second method. 
Here you need a normal wifi connection beween 2 devices and you need the mac adress of them. 
then you configure nexmon to the mac addresses and start nexutils. atfter that you should be able to receive CSI values by listening to the UDP datagramms at the nexmon device with tcpdump or wireshark, which contains the csi values of the received layer2 frames of the sniffed commmunication.

bolbolsaj, 2020-12-30:
My apologies 
which channel/bandwidth we should use? 
My pi 4 is in channel 7.
I use this code on terminal:
mcp -c 07/80 -C 1 -N 1 -m macaddress

The error:
Invalid chanspec

Can it be caused by channel number?

tweigel-dev, 2020-12-30:
@bolbolsaj 
currently I have no reachable  environment to validate my 2 hypothese but :
1. are you sure to use 07 instead 7 at the channel ? `mcp -c 7/80 -C 1 -N 1 -m macaddres`
2. you have to configure the channel and bandwith to the communication you want to gather CSI. therefore I recoment to check which channel and bandwith is used at the the communication and try to configure mpc. Because I think there is no 80MHz bandwith at 2.4 GHz channels.

bolbolsaj, 2020-12-30:
@tweigel-dev no. i just checked the issues for my problem and this one seems close to my problem.
i use 7/80. the same error happened. i checked other issues and in #147 they used 36/80.

tweigel-dev, 2020-12-30:
36 is a 5GHz Channel . I think 2.4 GHz don't allow to usw 80MHz

matthiasseemoo, 2020-12-30:
Technically, it works. Feel free to read our reactive jamming paper for details.

yjxb1, 2023-09-20:
> 我想使用 CSI 值进行活动识别。我在两个覆盆子之间做一些活动（走路，跌倒,...）并收集CSI值。

Hello, has your problem been solved? I do a similar job to you. I have some questions that I would like to ask you.