seemoo-lab/nexmon_csi #164: How to connect Asus-RTAC86U to Access point wirelessly?

adl94, 2020-12-21:
Greetings, 

I have two Asus-RTAC86U routers one is a receiver that has nexmon firmware and the other is an access point. I have configured the router as an access point but not able to connect it with the other router (receiver). 

I am attaching the architecture of my network. 

[image]

zeroby0, 2020-12-21:
What is C1 and C2?

You may be able to plug in a USB WiFi adapter to the RT-AC86U with Nexmon, but the real answer is to just get an Ethernet cable. I think I wasted a month on trying to get WiFi adapters working with Nexmon installed, and it didn't even work in the end. It's especially difficult because wpa_supplicant is needed to connect to WPA encrypted WiFis.

There's a 50 feet Amazon basics cable available. If you connect the Nexmon router, and your computer, to the Access point router, you would have a pretty neat setup. And 50 feet is probably at the edge of WiFi coverage anyway. You could even power your router over Ethernet via PoE (Additional equipment worth ~20$ is needed).

#42 has more info.

adl94, 2020-12-21:
> What is C1 and C2?
> 
> You may be able to plug in a USB WiFi adapter to the RT-AC86U with Nexmon, but the real answer is to just get an Ethernet cable. I think I wasted a month on trying to get WiFi adapters working with Nexmon installed, and it didn't even work in the end. It's especially difficult because wpa_supplicant is needed to connect to WPA encrypted WiFis.
> 
> There's a 50 feet Amazon basics cable available. If you connect the Nexmon router, and your computer, to the Access point router, you would have a pretty neat setup. And 50 feet is probably at the edge of WiFi coverage anyway. You could even power your router over Ethernet via PoE (Additional equipment worth ~20$ is needed).
> 
> #42 has more info, but if you do somehow get it working, please let me know :)

Thank you so much for your reply. C1 and C2 are the two laptops. C1 connects to  Router 1(Access Point) via LAN cable and C2 connects to Router 2 (Receiver) also via LAN cable.  I want to send packets from C1 and capture them on C2. 

But not able to figure out how routers would connect?