seemoo-lab/nexmon_csi #165: Is channel 36 needed for Raspberry pi (bcm43455c0) CSI collection?

Tom-P-Michaelis, 2020-12-30:
My current setup has a TP-LINK CPE510 5GHz 802.11n access point. This router seems to only transmit on channels 100-140. I am unsure if this will cause a problem as, despite not being mentioned in the read me, some users suggest channel 36 is needed, which seems to be confirmed by https://github.com/seemoo-lab/nexmon_csi/blob/master/src/regulations.c :

[image]

Do I need to buy new hardware or is there a way around this?

Thanks in advance!

zeroby0, 2020-12-30:
### Short answer

Buy [TPlink Archer C-20 (AC750)](https://www.amazon.in/TP-Link-Archer-C20-Wireless-Router/dp/B0759QMF85) and install OpenWRT. You can transmit in any channel you desire, and I've confirmed that it works perfectly with Raspberry Pi. Other models may not work; my TPlink C-80 AC1950 doesn't work, for example.

### Long answer

This may or may not work. I think I heard some on the issues that only channel 36 is known to work perfectly with Nexmon CSI. I've tried other channels by modifying the regulations.c file, but channel 36 is what works best and works consistently. You can get a chanspec by setting it with `nexutil -k36/80` and then running `nexutil -k` to see the chanspec `0xe02a`. You can also ignore that part and use the `CHXXMHZ_CHSPEC()` macros to add channels to your regulations file. The exact way to use the macro can be found in the issues, but let me know if you need more help there.

And as I always advice, please go through the previously created issues, they have answers to 95% of the problems you may encounter when using Nexmon CSI. Take a look at the paper (Link in Readme) too, it's really helpful in understanding how the firmware works.

### Alternative answer?

Maybe create an access point with another Raspberry Pi? You can also inject frames with a Raspberry Pi using the main Nexmon framework (I have no experience in doing this), and it will let you set the bandwidth and framerate, etc. You can also create a WiFi hotspot with you mobile and see if that works.

Tom-P-Michaelis, 2021-01-01:
Thanks a lot! very helpful

closed 2021-01-01 by Tom-P-Michaelis

stepanmazokha, 2021-02-02:
@zeroby0 thank you for your advice to consider the Archer C20 router as an emitter. 

Have you tried emitting frames on 2.4 GHz channels?

I have tried emitting probe requests on 1-11 channels, but they seem to always be captured using 11b standard (CCK modulation) which isn't supported by Nexmon (I'm using Alfa AWUS036ACH sniffer for capturing probes to check if transmission works, and it supports a/b/g/n+ standards).

Perhaps you could suggest any steps to fix this issue and force transmission using OFDM modulation?