seemoo-lab/nexmon_csi #166: How to  inject a frame from  Asus router with only a single spatial stream

Cyrbaby, 2020-12-30:
Greetings,
I read the  publication “ Free Your CSI: A Channel State Information Extraction Platform For Modern Wi-Fi Chipsets.” The publication mention “In the first (top) we inject a frame from another Asus router with a single spatial stream (NSS = 1).
In this case, the router transmits the same data Space Time Block Coded (STBC) over four antennas. ”

How can I accomplish this step using Asus RT-AC86U? Since I want to use nexmon CSI Extractor test multi-target localization performance, I don't want different target is separated by spatial stream.

Can I control the number of spatial stream in  transmitter?

Thanks in advance.

matthiasseemoo, 2020-12-30:
You can define the number of spatial streams to use during transmissions: https://github.com/seemoo-lab/nexmon/blob/master/patches/include/rates.h#L36 <https://github.com/seemoo-lab/nexmon/blob/master/patches/include/rates.h#L36>

The rate is a parameter of the sendframe function: https://github.com/seemoo-lab/nexmon/blob/master/patches/bcm4356/7_35_101_5_sta/nexmon/src/sendframe.c#L48 <https://github.com/seemoo-lab/nexmon/blob/master/patches/bcm4356/7_35_101_5_sta/nexmon/src/sendframe.c#L48>

Cyrbaby, 2020-12-31:
> You can define the number of spatial streams to use during transmissions: https://github.com/seemoo-lab/nexmon/blob/master/patches/include/rates.h#L36 [...]

Thank you for the hint. NEXMON is a really useful tool for frame injection.However, the devices NEXMON support don't include the ASUS router, can I control the number of spatial stream in  Asus RT-AC86U?

matthiasseemoo, 2021-01-01:
> … <x-msg://15/#>

In the end, it would be no problem to implement it. Its always just a question of time.

Cyrbaby, 2021-01-04:
Thanks for the reply

yujianyuanhaha, 2021-01-19:
@Cyrbaby @matthiasseemoo Well, I think simply a SMA terminator will do to "block" 3 antennas and leave only 1 antennas. For ASUS router, you need to take off all 3 attennas and replace them with SMA terminator.  In this way, there is only 1 TX attennas.

[image]

Cyrbaby, 2021-01-21:
> @Cyrbaby @matthiasseemoo Well, I think simply a SMA terminator will do to "block" 3 antennas and leave only 1 antennas. For ASUS router, you need to take off all 3 antennas and replace them with SMA terminator. In this way, there is only 1 TX antennas.
> 
> [image]

Thanks for the answer！But I just want use 4 TX antennas all，and these antennas  transmit the same spatial stream.

matthiasseemoo, 2021-01-21:
You should only use terminators on receive antennas. On transmit antennas, they will likely reflect most of the transmission power back into your transmitter and might thereby damage your hardware. In rates.h we have some macros to define the number of spatial streams (NSS) [1]. That does not necessarily mean that only one antenna is used for transmission. One stream can also be transmitted using multiple antennas using STBC. Then you can observe the effects in Figure 7 of our Free your CSI paper.

[1] https://github.com/seemoo-lab/nexmon/blob/master/patches/include/rates.h#L36 <https://github.com/seemoo-lab/nexmon/blob/master/patches/include/rates.h#L36>

Cyrbaby, 2021-01-23:
> You should only use terminators on receive antennas. On transmit antennas, they will likely reflect most of the transmission power back into your transmitter and might thereby damage your hardware. [...]

Thanks for the reply！I am researching the related code.

Deltaaaaaaaaaa, 2021-06-03:
> Thanks for the answer！But I just want use 4 TX antennas all，and these antennas transmit the same spatial stream.

Hi, if you are still working on this, maybe you can transmit on one antenna(SMA connector on AC86U), and apply a power splitter to the SMA connector, split the signal to 4 antennas.
I hope it would help!

yujianyuanhaha, 2021-06-18:
@Deltaaaaaaaaaa Where can we get SMA female-to-female cables? It is safe to do so?

Arsering, 2021-07-30:
@Cyrbaby Hi, Are you able to inject frame on ASUS RT-AC86U?