seemoo-lab/nexmon_csi #168: Uncorrect source mac address

ENB213-xiaoh, 2021-01-06:
Hi, all. First, thank you for building this powerful tool. I do have one problem here. This tool is functional on my Pi 4B. But when I look into the pcap file. I found that the shown source mac address is not right. The last byte is changed. For example, the last byte of my AP is 0xa3, but the file shows 0xab or 0xb0. The first five bytes are correct. So, I'm wondering if its my fault or maybe I need to change the base code to recover it. Please kindly advise me. Thanks.

zeroby0, 2021-01-06:
Your AP has at least 3 Mac IDs:
* The Ethernet Mac for the WAN port
* The 2.4 GHz Radio Mac ID
* The 5 GHz Radio Mac ID

For all of these, usually only the last byte is different. Maybe you're seeing the 5 GHz Mac ID, and what you're expecting to see is the WAN Ethernet Mac ID (Usually shown on router's web ui)?

You can see the 5 GHz Radio's Mac ID on the routers Web UI, or by running `ifconfig` on a wireless device connected to it, and looking at the BSSID, or by running `iw list wlan0 scan` on a Raspberry Pi before loading Nexmon.

ENB213-xiaoh, 2021-01-06:
Thank you for reply. Yes, im getting from 5GHz. Is there any way to get around this problem? So, to get the full mac address.

closed 2021-01-07 by ENB213-xiaoh