seemoo-lab/nexmon_csi #170: Collect CSI from multiple APs simultaneously

yangtat, 2021-01-09:
Greetings, Now I want to try using multiple APs
If I want to capture CSI from multiple APs, should I use `makecsiparams -m mac1 mac2……`？Add multiple AP MAC addresses after `-m`?
If I collect CSI from multiple APs, how do I split them up using MATLAB code?

yujianyuanhaha, 2021-01-10:
it is possible to do so. @yangtat 
The idea is using the command line tool `tshark` can help you extract the Mac address of each package in the `pcap` file, and then write a simple matlab or python code to split the package by the Mac address.

yangtat, 2021-01-11:
@yujianyuanhaha Thank you for your advice. I found that the author had read the data of PCAP.
put`MAC=dec2hex(payloadbytes(47:52))` behind `payloadbytes = typecast(payload,'uint8');` ，and we can get the mac

closed 2021-01-11 by yangtat