seemoo-lab/nexmon_csi #171: execute source setup_env.sh  but“Platform not supported!”

Cyrbaby, 2021-01-13:
Greetings,
I follow the getting started guide for bcm4366c0 in my ubuntu 18.04.5 LTS 32bits system.After I execute source setup_env.sh, I only see “Platform not supported!” .

May I ask the reason?

execute uname -s I can see "Linux"
execute unmae -m I can see "i686"

Thanks in advance.

zeroby0, 2021-01-13:
I have pre-compiled binaries of asus rt-ac86u, and steps to install them here: https://github.com/nexmonster/nexmon_csi/discussions/2

You can use them to get CSI while you get this compile error sorted out.

yujianyuanhaha, 2021-01-19:
@Cyrbaby  Follow @zeroby0 's way should be the easiest way. Simply copy and paste the `dhd.ko` `nexutil` `tcpdump` via `scp` command

Cyrbaby, 2021-01-21:
> I have pre-compiled binaries of asus rt-ac86u, and steps to install them here: [nexmonster#2](https://github.com/nexmonster/nexmon_csi/discussions/2)
> 
> You can use them to get CSI while you get this compile error sorted out.

Thanks for your advice！It is really useful

Cyrbaby, 2021-01-21:
> @Cyrbaby Follow @zeroby0 's way should be the easiest way. Simply copy and paste the `dhd.ko` `nexutil` `tcpdump` to `/jffs/` folder, via `scp` command.

Thanks for the answer，you‘re right！’