seemoo-lab/nexmon_csi #172: RaspberryPi4 8GB faild to follow the stepb on cm43455c0

sowai00, 2021-01-14:
how to fix it ?

Step7: Check if /usr/lib/arm-linux-gnueabihf/libmpfr.so.4 exists, if not, compile it from source: cd buildtools/mpfr-3.1.4, ./configure, make, make install, ln -s /usr/local/lib/libmpfr.so /usr/lib/arm-linux-gnueabihf/libmpfr.so.4

Now : usr/lib/arm-linux-gnueabihf/libmpfr.so.4 is not exists, 

root@raspberrypi:/home/pi/nexmon/buildtools/mpfr-3.1.4# ./configure
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether to enable maintainer-specific portions of Makefiles... yes
checking build system type... armv7l-unknown-linux-gnueabihf
checking host system type... armv7l-unknown-linux-gnueabihf
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for a sed that does not truncate output... /usr/bin/sed
checking for gcc... /home/pi/nexmon/buildtools/gcc-arm-none-eabi-5_4-2016q2-linux-armv7l/bin/arm-none-eabi-
checking whether the C compiler works... no
configure: error: in `/home/pi/nexmon/buildtools/mpfr-3.1.4':
configure: error: C compiler cannot create executables
See `config.log' for more details
root@raspberrypi:/home/pi/nexmon/buildtools/mpfr-3.1.4# 

the /home/pi/nexmon/buildtools/mpfr-3.1.4/config.log 
[config.log](https://github.com/seemoo-lab/nexmon_csi/files/5813460/config.log)

sowai00, 2021-01-14:
/usr/lib/arm-linux-gnueabihf/libmpfr.so.6 is exists

sowai00, 2021-01-14:
root@raspberrypi:/home/pi/nexmon/buildtools/mpfr-3.1.4# make
 cd . && /bin/bash /home/pi/nexmon/buildtools/mpfr-3.1.4/missing automake-1.15 --gnu
/home/pi/nexmon/buildtools/mpfr-3.1.4/missing: line 81: automake-1.15: command not found
WARNING: 'automake-1.15' is missing on your system.
         You should only need it if you modified 'Makefile.am' or
         'configure.ac' or m4 files included by 'configure.ac'.
         The 'automake' program is part of the GNU Automake package:
         <http://www.gnu.org/software/automake>
         It also requires GNU Autoconf, GNU m4 and Perl in order to run:
         <http://www.gnu.org/software/autoconf>
         <http://www.gnu.org/software/m4/>
         <http://www.perl.org/>
make: *** [Makefile:397: Makefile.in] Error 1
root@raspberrypi:/home/pi/nexmon/buildtools/mpfr-3.1.4#

sowai00, 2021-01-14:
/usr/share
aclocal-1.16 is there

zeroby0, 2021-01-14:
Try installing automake, and run `autoreconf -f -i` before ./configure.

I have a script to do the install here: https://github.com/nexmonster/nexmon_csi/tree/pi-5.4.51