seemoo-lab/nexmon_csi #174: Intermittent CSI collection on Raspberry Pi

Tom-P-Michaelis, 2021-01-16:
I'm trying to collect CSI on a Raspberry pi 4. When using tcpdump the collection stops for around 2 seconds roughly every 6 seconds which is stopping me using the CSI for my intended purpose. 

My router is a TP-Link CP510 that is transmitting on channel 36 at 40Hz using Openwrt. I am generating traffic using Iperf between 2 devices, although the problem also persists when I use  the ping command from one device.  

During the periods when CSI is not collected there are sometimes some packets of length 274 (as opposed to the 512 normally collected).  

I can't seem to find any similar issues from this forum, can anyone help me with this? Thanks in advance - as always super appreciated!

zeroby0, 2021-01-17:
Are you sure tcpdump is actually pausing collection? You can verify by looking at timestamps in pcap record header.

Tcpdump caches packets in a buffer, and then processes them in batches for efficiency. This could look like tcpdump is pausing and resuming.

Tom-P-Michaelis, 2021-01-17:
[image]

Unfortunately CSI is in fact intermittent, even when using the -U parameter in tcpdump.

yujianyuanhaha, 2021-03-02:
@Tom-P-Michaelis set up iperf3 communication between a client and the TX router can bring up streaming CSI

saydeking, 2021-04-22:
Hi! Thank you for your work on this tool! We unfortunately are having a similar problem where after filtering to the Mac address of our TCP-Link router, we collect packets very slowly even when setting the ping rate to be very low with a high volume of pings. Here's a look at our pcap file, you can see the time stamps of the packets. 
[image]