seemoo-lab/nexmon_csi #175: Pi4 CSI of multiple spatial streams simply a replica

m-tahsin, 2021-01-19:
Hello,

Firstly, thank you for the tool and the great community building around it. I have learned a lot!

My setup:
Rx: Pi4 running nexmon_csi. I have configured it with makecsiparams -c 36/20 -C 1 -N 15
Tx: ASUS RT-AC86U transmitting beacon frames every 100ms on channel 36 with bandwidth 20MHz. No nexmon installed on Tx.
Additionally: My laptop connects to Tx and sends ping requests.

The problem:
I am able to capture CSI at the Rx. For each transmission I get 4 UDP packets in the .pcap (note that -N 15 corresponds to 0b1111), but the CSI in each of the 4 packets is exactly the same. I would expect the different spatial streams to have different CSI. More specifically, I would expect the channel between the different transmit antennas and receive antennas to be different. Note that spatial stream is not necessarily equal to the no. of antennas all the time. I know that these UDP packets correspond to the same packet with different spatial streams because the "Core and Spatial Stream" byte changes as 00, 08, 10, 18 (the 3 lowest bits are all 0, which indicate the single core; the next 3 lowest bits are 000, 001, 010, 011, which indicate the change in spatial stream).

I have looked at the CSI from beacon frame packets from the Tx. I have looked at the ping replies from Tx to my laptop. I even capture CSI from the original ping from my laptop, and it shows 4 replicas of the CSI even though my laptop does not have 4 antennas. The time stamps of the spatial stream replicas are different by about 0.1ms as seen in Wireshark. The only difference I see is that sometimes there are an additional 4 bytes at the end of the packet, which Wireshark labels as "Frame check sequence". In the example.pcap file provided I see that the different spatial streams have different CSI (there are technically 2 spatial streams as the example shows 2x2:2, but the CSI is different for all four channels, which suggests that the CSI is not for spatial streams but for transmit to receive antenna channels).

What is going on? Any help would be very much appreciated @matthiasseemoo @jlinktu

zeroby0, 2021-01-20:
Can the pi resolve multiple spatial streams with 1 antenna? I think it's just showing the aggregate of all spatial streams on all of them.

m-tahsin, 2021-01-20:
A heartfelt thank you to @zeroby0. I used your forked repo to get CSI collection on the pi working :)

Regarding the spatial streams, you are right. Nss <= rank(H), where H is the channel matrix between antennas. For example, 2 transmitters and 2 antennas will form a 2x2 channel matrix, which has max(rank(H))=2. So a 2x2 system can support a maximum of 2 spatial streams. Each spatial stream is actually an independent stream of data.

What gets me confused is that in example.pcap we have 2 cores and 2 spatial streams. This gives 4 CSI packets. But the 4 CSI are all different. As mentioned above, in a 2x2 system we can have a maximum of 2 spatial streams. So, if what you suggested is true and there is some aggregation, we would have to see 2 of those CSI being replicas of each other... Anyways, I'm not sure it is technically correct to say CSI of a 'spatial stream'. From example.pcap, and my general understanding, the CSI from each core (Rx antenna) and each 'spatial stream' (a.k.a Tx antenna), should be unique but correlated.

Does my understanding and reasoning make sense? If I were to collect CSI on the router instead, do you think I would get the same effect? The channel is reciprocal, so I would think yes, but my gut instinct says that the CSI will be different. I want to collect CSI on the Pi though but will do so on the router as a last resort.

I'm not very good at reading the underlying code. Does anyone know what *exactly* the -N parameters does?

P.S. The CSI we get is the channel matrix * other processing matrices, one of which is the spatial mapping matrix (SMM). However, the effect of the channel should appear in the resulting 4x1 CSI that we get, and we should see differences obtained, even if they are highly correlated. This is true unless the channel between the Tx antennas and Rx is the same, which will only happen for anechoic chamber measurements where the Tx and Rx are facing each other or if you connect conductors of same length between the Tx and Rx.

zeroby0, 2021-01-22:
Is this the example.pcap you're referring to? https://github.com/seemoo-lab/nexmon_csi/blob/master/utils/matlab/example.pcap

I think it's created from Nexus 6P which has multiple antennas.This line in csireader.m shows that the chip is bcm4358:
https://github.com/seemoo-lab/nexmon_csi/blob/ba99ce12a6a42d7e4ec75e6f8ace8f610ed2eb60/utils/matlab/csireader.m#L12

I would be very interested to see how the CSI will be different when collecting from the router! My gut says they should be slightly different because the antennas are spaced apart, but I don't know if they will be different enough for us to notice.

Thank you for the attached theory, it helped me understand rfcomms a little better :). Glad you found my fork useful, I really struggled with nexmon_csi when I got started, so I'm trying to make things easier now :).

DennisHerell, 2021-02-03:
Hi! I also got the same result of replicated CSI with RPi 3B+

From another issue #160 it seem like we can't extract csi from multiple spatial stream with only 1 core. In that issue, nexus 5 also got a replicated csi while nexus 6p with 2 cores manage to get different result. However, @m-tahsin  your explanation make sense to me, and there's another mention in  #112 that we should be able to estimate CSI from all 4 transmit antenna even with RPi which only have 1 antenna. 

I'm new to this field of study, so I might not be able to help. However, from what I know the -N parameter send ioctl message to 
the assembly code in d11 as [NSSMASK] as can be seen in https://github.com/seemoo-lab/nexmon_csi/blob/master/src/csi.ucode.bcm43455c0.7_45_189.patch#L326 (For anyone who's trying to read the assembly code, I found this very helpful https://github.com/fullstory/b43-asm/tree/master/assembler)

Do you have any update regarding extracting CSI from multiple spatial stream using RPi?

P.S. I also want to thank @zeroby0 for your forked repo, it has been very helpful in getting started on collecting CSI

m-tahsin, 2021-02-03:
@DennisHerell Thank you for your follow up and the links to the other issues. I have tried without success to get multiple spatial streams with a single receive antenna. When comparing with the other responses (e.g. 4x4:4 ASUS Rx/Tx and 2x2:2 Nexus) it seems there is a bug somewhere when considering a single antenna receiver.

This is a major drawback for me, so I am moving to another solution using SDR. I will keep my eye on this repo for any updates.

m-tahsin, 2021-03-19:
I would like to correct a previous misconception. For a single antenna receiver system there can be no more than 1 spatial stream. If the transmitter has multiple antennas, it will send the same stream of data on each of them, except that it will add a delay between transmissions on antennas (about 200ns) to avoid unwanted transmit beam forming. These identical transmissions from multiple antennas look like signal reflections to the receiver. In effect, the receiver is seeing a single stream of data and will have a single CSI vector regardless of the number of transmit antennas. 

With this in mind, there is no bug and the behaviour is expected. Now closing the Issue.

closed 2021-03-19 by m-tahsin

zeroby0, 2021-03-20:
@m-tahsin If you have them, can you please add links to how you found out about this? I would like to read it as well :)

matthiasseemoo, 2021-03-20:
Next generation wireless lans is quite a good book to learn about the lower wifi layers:
https://www.cambridge.org/de/academic/subjects/engineering/wireless-communications/next-generation-wireless-lans-80211n-and-80211ac-2nd-edition?format=HB <https://www.cambridge.org/de/academic/subjects/engineering/wireless-communications/next-generation-wireless-lans-80211n-and-80211ac-2nd-edition?format=HB>

zeroby0, 2021-03-21:
@matthiasseemoo 
Thanks for suggesting this book! I bought it and read a chapter, and it's very nice!

m-tahsin, 2021-03-25:
@zeroby0 Perahia and Stacey's book linked above is a classic. Matlab uses it as a reference to implement packet detection in some of their packages, for example. You can refer to Chapter 4 there. You can certainly do worse that looking through Cho's book as well:

Y. S. Cho, J. Kim, W. Y. Yang, and C. G. Kang. MIMO OFDM Wireless Communications with MATLAB. John Wiley & Sons, 2010.

haochenku, 2021-07-09:
Hi @m-tahsin @zeroby0 @DennisHerell , 802.11ac will send duplicated LTF frames to help Rx channel estimation. So at this stage, there is no decision on number of steams yet. Based on, the Rx even with 1 core should be able to estimate its channel to each Tx antennas. I am suspecting there might be some implementation logic on the current code stopping the estimation of CSI over multiple Tx ant.