seemoo-lab/nexmon_csi #176: How should I choose the parameters of the attenuator

Cyrbaby, 2021-01-24:
Greetings,
I consider use the attenuator and splitter to test the phase difference between the antenna，can I use the attenuator whose power is 2W，and attenuator is 30dB？

Thanks in advance

quantumhub, 2021-02-08:
Wif TxPower is less than 30dBm (1W). So, 2W is fine.
30dB attenuation is about the pathloss at about 1m. It should not damage your receiver.

Cyrbaby, 2021-02-25:
> Wif TxPower is less than 30dBm (1W). So, 2W is fine.
> 30dB attenuation is about the pathloss at about 1m. It should not damage your receiver.

Thank you for your help， it is really help to me

yujianyuanhaha, 2021-07-02:
@Cyrbaby may I know where to purchase the attenuator and splitter, I found RP SMA is hard to find on market.

zeroby0, 2021-07-03:
> I found RP SMA is hard to find on market.

Yup, that's why they make them RP XD

Maybe check https://www.rtl-sdr.com/ and the re-seller stores listed there? The reseller stores is where I usually find the obscure components.