seemoo-lab/nexmon_csi #177: Inconsistent CSI collection in Raspberry Pi 4B, kernel 5.4 and firmware 7.45.189

maurovm, 2021-02-02:
I want to collect CSI packets at regular intervals. However, I found several bottlenecks that I hope you can help me.

Wifi router:
---

- Asus RT-AC86U
- Configured to run 5 Ghz, channel 36 bandwidth 80 Mhz

Traffic generation:
---

The computer that generates packets, let's call it "PI_TX" here for simplicity, has the following specs: 

- Raspberry Pi 4 Model B with 4GB ram
- Rasbian  OS Buster lite, originally downloaded image "2021-01-11-raspios-buster-armhf-lite.img". 
- Updated to the latest packages (as per the instructions) to run kernel 5.4.83-v71+
- Installed the nexmon software only. Note that I didn't install the software from the "nexmon_csi" repository.
- I used the 7_45_189 firmware from the folder "patches/bcm43455c0/7_45_189/nexmon/"
- I configured the brcm kernel module (from "patches/bcm43455c0/7_45_189/nexmon/brcmfmac_5.4.y-nexmon/brcmfmac.ko") to autoload at boot time. 
- The machine is configured to join the wifi network at boot time. It runs wpa_supplicant, dhcpcd and avahi daemons.

I had not have any issues following the instructions and install the required software. After rebooting, the computer loads the correct module, as the output ofd the following command:

```shell
$ dmesg | grep -i brcmfmac
```

Show the module version 7.45.189:

```
...
[    5.185108] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Feb  2 2021 11:51:27 version 7.45.189 (nexmon.org: 2.2.2-329-g5dbc-2) FWID 01-e1db26e2
...
```

The machine can successfully connect to the wifi router, ping any machine in the network, browse the internet, etc.

The output of the "iw dev" command is (I replaced my MAC address and name of my network):

```
phy#0
	Unnamed/non-netdev interface
		wdev 0x2
		addr MY_MAC_ADDRESS
		type P2P-device
		txpower 31.00 dBm
	Interface wlan0
		ifindex 4
		wdev 0x1
		addr MY_WIFI_ROUTER_ADDRESS
		ssid MY_NETWORK_SSID
		type managed
		channel 36 (5180 MHz), width: 80 MHz, center1: 5210 MHz
		txpower 31.00 dBm
```

Which shows the correct channel and bandwidth.

To generate traffic, I ping either the wifi router or another machine in the network with:

```shell
$ sudo ping -i 0.5 IP_OF_TARGET_MACHINE
```
or:

```shell
iperf3 -i 0.5 -u -t 0 -c IP_OF_TARGET_MACHINE
```

Both of the commands above work
[... 166 lines cut ...]
005113.136 
   sp+0 00000025 00242b78 ffffe800 00000001
005113.136   sp+10 0025d560 000040d7 002363a8 0000000f

005113.136 sp+14 000040d7
005113.136 sp+44 001cd627
005113.136 sp+54 001a34f7
005113.136 sp+64 001a6f05
005113.136 sp+a4 001a72bd
005113.136 sp+cc 001ab4f5
005113.136 sp+f0 0000ffff
005113.136 sp+12c 00025b4b
005113.136 sp+144 001b6b07
005113.136 sp+184 001c1c47
005113.136 sp+1bc 001c87f1
005113.136 sp+1e4 001a6939
005113.136 sp+1f4 001a6975
005113.136 sp+204 0019a4d9
005113.136 sp+20c 0019aba9
005113.136 sp+21c 00199c47
```

This problems occurs very often. The only way to solve it is to reboot the machine.

Thank you for all your help

maurovm, 2021-02-02:
Adding to the description of my problem. From time to time, CSI collection could not even start. When executing "tcpdump", I get the following kernel error (shown by dmesg):

```
[  156.362831] device wlan0 entered promiscuous mode
[  158.872825] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[  158.876305] ieee80211 phy0: _brcmf_set_multicast_list: Setting mcast_list failed, -110
[  161.432816] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[  161.436293] ieee80211 phy0: _brcmf_set_multicast_list: Setting allmulti failed, -110
[  163.992809] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[  163.996285] ieee80211 phy0: _brcmf_set_multicast_list: Setting BRCMF_C_SET_PROMISC failed, -110
[  166.552800] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[  169.112819] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
```

Any ideas why it might happened?

maurovm, 2021-02-02:
Other times, I can collect CSI for a few minutes. Afterwards, it crashes with the following kernel error (as reported by dmesg):

```
[  149.307285] device wlan0 entered promiscuous mode
[  188.962074] ieee80211 phy0: brcmf_fw_crashed: Firmware has halted or crashed
[  264.287049] device wlan0 left promiscuous mode
[  266.797301] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[  266.800779] brcmfmac: brcmf_sdio_checkdied: firmware trap in dongle
```

jlinktu, 2021-02-03:
Problem 1:
Not sure about this one. Btw. there is no need to use the nexmon firmware on PI_TX if you are just connecting to an AP and use `ping` or `iperf` to generate traffic.

Problem 2:
This is correct. Please have a second look at the [readme](https://github.com/seemoo-lab/nexmon_csi).

Problem 3:
Traffic generated by `ping` or `iperf` does not necessarily translate to constant Wi-Fi traffic.
Have a look at the `tcpdump` manual for explanations of those numbers and options to control them.

Problem 4:
This might be related to AMSDUs, please try [this fix](https://github.com/seemoo-lab/nexmon_csi/pull/46/commits/82161c4199d112aedae5fb4ec2d3f191ff387e4c) and report back.

maurovm, 2021-02-03:
Dear  Jakob,

Thanks for the suggestions, they helped a lot. Please see below my answers: 

> Problem 1:
> Not sure about this one. Btw. there is no need to use the nexmon firmware on PI_TX if you are just connecting to an AP and use `ping` or `iperf` to generate traffic.
> 

The problem is that my MacBook does not tag frames with 0x88. Therefore, if I do:

```shell
$ makecsiparams -c 36/80 -C 1 -N 1 -m MAC_ADDRESS
```
I get traffic from the laptop. However, I need to add "-b 0x88" if I use a Raspberry Pi to generate traffic.

Problem solved.

> Problem 2:
> This is correct. Please have a second look at the [readme](https://github.com/seemoo-lab/nexmon_csi).

Thank you, problem solved.

> Problem 3:
> Traffic generated by `ping` or `iperf` does not necessarily translate to constant Wi-Fi traffic.
> Have a look at the `tcpdump` manual for explanations of those numbers and options to control them.
> 

Got it, problem solved.

> Problem 4:
> This might be related to AMSDUs, please try [this fix](https://github.com/seemoo-lab/nexmon_csi/pull/46/commits/82161c4199d112aedae5fb4ec2d3f191ff387e4c) and report back.

The fix you pointed out (by mzakharo) worked really well in my setup. I am now able to collect CSI data more consistently. I started/stoped CSI collection several times and left it for 30 minutes without any issues so far. Dropped packages are far less than what I had before too, which is wonderful.  However, I noticed a kernel error reporting an incorrect file "brcmfmac43455-sdio.raspberrypi,4-model-b.txt":

```
[    4.784266] brcmfmac: F1 signature read @0x18000000=0x15264345
[    4.797892] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    4.798461] usbcore: registered new interface driver brcmfmac
[    4.826801] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43455-sdio.raspberrypi,4-model-b.txt failed with error -2
[    5.213790] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    5.224916] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Feb  3 2021 11:06:42 version 7.45.189 (nexmon.org/csi: 55dc-2)
[    5.270728] ieee80211 phy0: brcmf_construct_chaninfo: Ignoring unexpected firmware channel 106
```

The above error does not seem to affect data collection, but I am not sure. Do you have some pointers on how to hunt for what might be the causes?

jlinktu, 2021-02-03:
That is the `brcmfmac` driver complaining about a missing nvram config file named `brcmfmac43455-sdio.raspberrypi,4-model-b.txt` in `/lib/firmware/brcm/`.
As you already figured it might be fine to just ignore it.
If you want to fix it, you can try renaming `/lib/firmware/brcm/brcmfmac43455-sdio.txt` to `/lib/firmware/brcm/brcmfmac43455-sdio.raspberrypi,4-model-b.txt`.

maurovm, 2021-02-03:
The file "/lib/firmware/brcm/brcmfmac43455-sdio.txt" looks to have important NVRAM flags. Reading the error logs and differences when I do and don't rename the file, it seems when it fails to find "brcmfmac43455-sdio.raspberrypi,4-model-b.txt" it retries with "brcmfmac43455-sdio.txt". Although I cannot verify this.

We are preparing to carry a lengthy research study. Do you have any pointers on how to begin debugging this?. One of my students will work on getting prepared the hardware to record data and I would like him to spend some time verifying this problem.

jlinktu, 2021-02-03:
It seems you can find the correct `brcmfmac43455-sdio.raspberrypi,4-model-b.txt` file here:
[https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/brcm](https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/brcm).

If no nvram file is loaded the firmware has fallback values that it uses instead.

maurovm, 2021-02-04:
Thanks Jakob for all your help, this really helps. I am now closing this issue

closed 2021-02-04 by maurovm