seemoo-lab/nexmon_csi #178: Injecting Frames on 2.4 Ghz

stepanmazokha, 2021-02-02:
Hi all,

Thank you for this fantastic library and your continuous work on the project!

My problem consists of multiple parts, so I'll describe my setup first:

1. My goal is to be able to inject probe requests on both 2.4 GHz & 5 GHz and later capture them using Raspberry Pi 4B+ using Nexmon CSI to analyze the CSI data.

2. I have tried injecting probe requests using both Raspberry Pi 4B+ (via Alfa AWUS036ACH using Scapy, running Kali Linux 4.19 Re4son kernel) and using Archer C20 (AC750 V5, running OpenWRT).

3. To verify which probe requests are being emitted by one of the aforementioned devices (sort of like "ground truth"), I am using [Alfa AWUS036ACH sniffer](https://www.amazon.com/gp/product/B00VEEBOPG).

4. Thanks to [this answer](https://github.com/seemoo-lab/nexmon_csi/issues/15#issuecomment-566725579) I am aware that Nexmon patches support only OFDM modulation (this means that standard 11b isn't supported as it's using CCK modulation).

Here is my issue:

Whichever method of emitting I use (both via Alfa AWUS036ACH and via Archer C20), if I emit probes on one of the 2.4 GHz channels (I tried 1-11), every probe is being received using 11b protocol version. Due to issue mentioned in (4) above, Nexmon CSI cannot intercept such probes.

To be clear, both Alfa sniffer & Archer support a/b/g/n standards and should be capable of dispatching probes using OFDM.

If I'm trying to emit probes on 5GHz channels, probes are captured by 11a standard, which by default uses OFDM.

So, my question:

**How is it possible to inject probe requests so that they would be transmitted on 2.4 GHz via OFDM? Is my hardware incapable of doing so? The Alfa sniffer regularly intercepts 11g probes coming from other devices.**

Additional question:

**Are there any plans to support CCK modulation by Nexmon patches? Is this possible?**

Thank you in advance!

jlinktu, 2021-02-03:
Are you using [this patch](https://github.com/seemoo-lab/nexmon/tree/master/patches/bcm43455c0/7_45_206/nexmon) to inject frames on the RPi?
Then you can set the interface in monitor mode and prepend a radiotap header with an ```IEEE80211_RADIOTAP_RATE``` field to control the transmission rate (see [injection.c](https://github.com/seemoo-lab/nexmon/blob/master/patches/bcm43455c0/7_45_206/nexmon/src/injection.c)). Have a look at [rates.h](https://github.com/seemoo-lab/nexmon/blob/master/patches/include/rates.h) for more details on the rate value. You could e.g. use ```RATES_RATE_6M | RATES_OVERRIDE_RATE``` to force 6Mbps OFDM encoding.

Another option would be to form and transmit the probe requests directly in the firmware and set the encoding there.

No, the CSI extracted by this patch are specific to OFDM.

codeceptsDE, 2021-02-03:
> Are there any plans to support CCK modulation by Nexmon patches? Is this possible?

@stepanmazokha Just chipping in: I don't think this is possible, as for CCK frames there **is** no CSI in the same sense as for OFDM. For OFDM, the radio does channel estimation for all 64 or so frequency bins it is using concurrently, which yields one complex value (amplitue & phase of one bin) for each. These 64 or so values make up your CSI. 
CCK would just be using one carrier frequency, so there would only be a single channel state value to estimate, which is much less interesting, basically just the attenuation.

Disclaimer: I might be wrong about any of this, so please take this with a grain of salt, I'm not really an expert in this matter.
Good luck with your OFDM injection!

stepanmazokha, 2021-02-03:
Wow, thank you so much for replying so quickly! 

@jlinktu I have not yet tried injecting probes using Nexmon patch -- had a plenty of issues installing it onto my RPi. I tried only via Scapy and Aireplay-ng. I will now try to attack installing Nexmon on Raspbian 5.4.83-v7l+ and see if it works. 

By any chance, do you have any sample code for injecting packets using this code? For now, I found this [example](https://github.com/seemoo-lab/nexmon/issues/400#issuecomment-625946764), but in case you could suggest anything in addition to that -- I would highly appreciate it!

@codeceptsDE thank you very much for your explanation! The insight into CCK inner workings is very insightful!! 

I'll post my updates on Nexmon injection work here as I get further down the road.

jlinktu, 2021-02-03:
You could:
1. install the patched firmware
2. create a monitor interface
3. form your packet and prepend a radiotap header using e.g. Scapy
4. send it to the interface

stepanmazokha, 2021-02-03:
@jlinktu This is exactly what I tried. But unfortunately, I couldn't capture probe requests generated using Scapy at all when using Nexmon-patched firmware. Here's what I've been using:

**Sender**: RPi 4B+ (Raspbian 5.4.83-v7l+) with `7_45_206` version of the Nexmon patch installed (this time I figured out how to do it). Monitor mode works (tested on multiple 2.4 GHz channels, including 7). I've set interface up to use channel 7 using `nexutil -k7` command, and made share that nexutil is in mode 2 (`nexutil -m2`).

**Receiver**: RPi 4B+ (Kali Linux, listening on Alfa AWUS036ACH sniffer, also on channel 7 in monitor mode).

Here's the python code I use to send probes (monitor interface is called mon0):

```
def send_probes(device, interval):
    dot11 = Dot11(addr1='ff:ff:ff:ff:ff:ff', addr2='11:11:11:11:11:11', addr3='ff:ff:ff:ff:ff:ff')
    probeReq = Dot11ProbeReq()
    dot11elt = Dot11Elt()

    frame = RadioTap()/dot11/probeReq/dot11elt    
    sendp(frame, iface=device, loop=1, inter=interval)

send_probes('mon0', 0.05)
```

Monitor mode on Alfa sniffer works 100%, because I was able to capture probes coming from nearby devices during testing. But nothing has been coming with MAC 11:11:11:11:11:11.

Could you please suggest further steps? Am I doing something wrong here?

jlinktu, 2021-02-03:
Steps to reproduce.
```
make install-firmware
sudo iw phy `iw dev wlan0 info | gawk '/wiphy/ {printf "phy" $2}'` interface add mon0 type monitor
sudo ifconfig mon0 up
nexutil -k7
sudo -E python tx.py
```
Contents of `tx.py`:
```
from scapy.all import *

def send_probe(device, interval):
    rt = RadioTap(len=9, present='Rate') # 8 bytes radiotap header, 1 byte Rate field
    rt.Rate = 6 # Mbps
    dot11 = Dot11(addr1='ff:ff:ff:ff:ff:ff', addr2='11:22:33:44:55:66', addr3='ff:ff:ff:ff:ff:ff')
    probeReq = Dot11ProbeReq()
    dot11elt = Dot11Elt(ID="SSID", info="testSSID")
    frame = rt/dot11/probeReq/dot11elt
    sendp(frame, iface=device, loop=1, inter=interval)

send_probe('mon0', 0.05)
```
Sniffer:
```
6.0 Mb/s 2442 MHz 11g -33dB signal -92dB noise Probe Request (testSSID)
```
Apparently scapy expects the rate value in Mbps and scales it to 500 kbps units when forming the radiotap header.

stepanmazokha, 2021-02-04:
@jlinktu it worked!!! My issue with python code has been the incorrect definition of the RadioTap header. HUGE THANK YOU!

closed 2021-02-04 by stepanmazokha