seemoo-lab/nexmon_csi #179: Compile nexutil for ac88u

quantumhub, 2021-02-05:
I follow instruction and have /jffs/dlarray_4366c0.bin loaded on ac88u successfully. 
However, the nexutil complied for ac86u won't work on ac88u.

New wifi driver seems to be working:
admin@RT-AC88U:/jffs# wl -i eth2 channel
No scan in progress.
current mac channel	36
target channel	157

But nexutil is not:
admin@RT-AC88U:/jffs# ./nexutil -Ieth2 -k
./nexutil: line 1: ELF�H@@X$
                                 @8@@@,,�I�IP-�NXX@X@: not found

Hey @zeroby0! 

### To answer your question...
With the fantastic guidance of @jlinktu, having compared two dhd.ko files, I got the GT-AC5300 working by using the patched dhd.ko originally meant for the ac86u. This is because we found that the dhd.ko of the GT-AC5300 contains firmware version 10.10.122.20 compiled in 2017 (the ac86u firmware is also 10.10.122.20 but compiled in 2019).

### My progress on trying to get the RT-AC88U to work...
Funnily enough I've been trying to get the RT-AC88U working and am facing the exact same error.
I again liaised with the generous @jlinktu on this and below is a summary of the progress we made:

**Jakob:**
> The wifi firmware itself (it is embedded inside the kernel module) should work as the RT-AC88U has the same wifi chip as the RT-AC86U.
> However, they have different main CPU architectures. The RT-AC86U has a BCM4906 which is a 64bit ARM Cortex-A53, whereas the RT-AC88U comes with a BCM4709C0 which is a 32bit ARM Cortex-A9.
> The dhd.ko provided in the repository is taken from an RT-AC86U and thus indeed incompatible with the RT-AC88U.
> However, if you are lucky, it might be possible to embed the same firmware inside a dhd.ko compatible with the RT-AC88U.
> On a RT-AC88U, running a default router firmware, search for the installed dhd.ko (e.g. with "find / -type f -name dhd.ko").
> Best would be to find a router firmware (if there is any) where the dhd.ko has the wifi firmware of version 10.10.122.20 embedded. (you can check the version with "strings dhd.ko | grep Version").
> If you find one you can send it to me so I can check if it is possible to embed the modified firmware inside the compatible kernel module.

**Me:**
Sent over the dhd.ko files corresponding to the following router versions:

- 10.10.122.47 (Firmware Version 3.0.0.4.384.21045)
- 10.10.122.303 (Firmware Version 3.0.0.4.385.20631)

**Jakob:**

> unfortunately the firmwares inside the two modules you sent me are smaller than the one of the RT-AC86U, making it not so easy to replace them.
> There is another thing you can try.
> Follow https://github.com/seemoo-lab/nexmon_csi#bcm4366c0 up to step 7. Instead of "make install-firmware ..." run "make dlarray_4366c0.bin".
> This should output a file "dlarray_4366c0.bin". Copy this file to your RT-AC88U (scp dlarray_4366c0.bin admin@<your routers address>:/jffs/).
> Try loading the original module with the modified firmware:
> ssh admin@<your routers address> "/sbin/rmmod dhd; /sbin/modprobe dhd firmware_path=/jffs/dlarray_4366c0.bin"
> If there's no error message you can verify the correct firmware was loaded by looking at the output of "dmesg" on the router.
> There should be a line containing "nexmon.org/csi".
> 
> However, I'm not sure if the module really handles the parameter.. but it is worth a try.

**Me:**
Attempted this and it reading the console read-out, it seemed to be successful. My assumption was then that I should run through the steps again, however I just got the same error on Step 7. 

It may be that I shouldn't have just gone through the steps again but instead have done something different, am not sure. So would recommend trying the same steps and seeing if you have more success than I did!

Additionally, I did only try Jakob's first idea with the two versions mentioned, so although unlikely to find one compatible with the RT-AC86U, it might be worth trying the other ones.

_Originally posted by @maxedbeech in https://github.com/seemoo-lab/nexmon_csi/issues/162#issuecomment-744024808_

jlinktu, 2021-02-05:
The AC88U and AC86U do not share common/compatible architectures.
The `nexutil` cross-compile instructions on the [readme](https://github.com/seemoo-lab/nexmon_csi/blob/master/README.md) target the AC86U's aarch64 architecture.
The target architecture for the AC88U is armv7-a. Thus, you need a different toolchain. You could try following the steps as in the  [readme](https://github.com/seemoo-lab/nexmon_csi/blob/master/README.md) and replace step 9 and 10 with:
9.
```
export AMCC=$(pwd)/am-toolchains/brcm-arm-hnd/crosstools-arm-gcc-5.3-linux-4.1-glibc-2.22-binutils-2.25/usr/bin/arm-buildroot-linux-gnueabi-
export LD_LIBRARY_PATH=$(pwd)/am-toolchains/brcm-arm-hnd/crosstools-arm-gcc-5.3-linux-4.1-glibc-2.22-binutils-2.25/usr/lib
```
10.
```
cd utilities/libnexio
${AMCC}gcc -c libnexio.c -o libnexio.o -DBUILD_ON_RPI
${AMCC}ar rcs libnexio.a libnexio.o
cd ../nexutil
echo "typedef uint32_t uint;" > types.h
sed -i 's/argp-extern/argp/' nexutil.c
${AMCC}gcc -static -o nexutil nexutil.c bcmwifi_channels.c b64-encode.c b64-decode.c -DBUILD_ON_RPI -DVERSION=0 -I. -I../libnexio -I../../patches/include -L../libnexio/ -lnexio  -mcpu=cortex-a9 -march=armv7-a
scp nexutil admin@<address of your rt-ac86u>:/jffs/nexutil
ssh admin@<address of your rt-ac86u> "/bin/chmod +x /jffs/nexutil"
```

If that does not work out start by searching for a toolchain compatible with the AC88U and go along as above with the matching toolchain.

quantumhub, 2021-02-08:
Dear @jlinktu ,

Thank you very much for your help! The above cross compiling works. And Now nexutil is runable in asus ac88u.
However, nexutil can't accept -s500 command on asus ac88u. Could anybody help? Thanks!

admin@RT-AC88U:/jffs# /sbin/rmmod dhd; /sbin/modprobe dhd firmware_path=/jffs/dlarray_4366c0.bin
admin@RT-AC88U:/jffs# wl -i eth2 up
admin@RT-AC88U:/jffs# wl -i eth2 radio on
admin@RT-AC88U:/jffs# ifconfig eth2 up
admin@RT-AC88U:/jffs# ./nexutil -Ieth2 -s500 -b -l34 -viuIB/wAAAQB8A6tAWncAAAAAAAAAAAAAAAAAAAAAAAAyAA==
__nex_driver_io: error ret=-1 errno=95
admin@RT-AC88U:/jffs# ./nexutil -Ieth2 -s500 -b -l34 -vOuAB/wAAAQB8A6tAWncAAAAAAAAAAAAAAAAAAAAAAAAyAA==
__nex_driver_io: error ret=-1 errno=95
admin@RT-AC88U:/jffs# ./nexutil -Ieth2 -k
chanspec: 0xd024, 36

jlinktu, 2021-02-08:
Do you get an output for `/jffs/nexutil -Ieth2 -V`?

quantumhub, 2021-02-08:
@jlinktu I have following output from nexutil -Ieth2 -V. 
I couldn't find a exact asus merlin firmware containing driver version 10.10.122.20. I am using asus merlin 384.3.

admin@RT-AC88U:/jffs# rmmod dhd
admin@RT-AC88U:/jffs# /sbin/modprobe dhd firmware_path=/jffs/dlarray_4366c0.bin
admin@RT-AC88U:/jffs# ./nexutil -Ieth2 -V
firmware 10.10.122.303 (r666429) FWID 01-3dd394f
vendorid 0x14e4
deviceid 0x43c5
radiorev 0x2103eb
chipnum 0xaa90
chiprev 0x4
chippackage 0x0
corerev 0x41
boardid 0x72f
boardvendor 0x14e4
boardrev P500
driverrev 0xa0a7a12
ucoderev 0x0
bus 0x0
phytype 0xb
phyrev 0x21
anarev 0x0
nvramrev 0x0

platform             | firmware                         | vendorid | deviceid | radiorev   | chipnum | chiprev | chippackage | corerev | boardid | boardvendor | boardrev | driverrev | ucoderev  | bus | phytype | phyrev | anarev | nvramrev
-------------------- | -------------------------------- | -------- | -------- | ---------- | ------- | ------- | ----------- | ------- | ------- | ----------- | -------- | --------- | --------- | --- | ------- | ------ | ------ | --------
unknown              | 10.10.122.303 (r666429) FWID 01-3dd394f |   0x14e4 |   0x43c5 |   0x2103eb |  0xaa90 |     0x4 |         0x0 |    0x41 |   0x72f |      0x14e4 |     P500 | 0xa0a7a12 |       0x0 | 0x0 |     0xb |   0x21 |    0x0 |      0x0

jlinktu, 2021-02-10:
You should see `10.10.122.20 (nexmon.org/csi)` as firmware version if the patched firmware was loaded correctly.
Thus, it seems the firmware was not loaded. Instead the default firmware, version `10.10.122.303` in your case, is loaded, which does not support an ioctl 500 and hence the error when trying to invoke it using `nexutil`.

quantumhub, 2021-02-15:
Thanks for letting me know that the patched firmware loading failed. 

Is it due to the difference in instruction sets used by ac88u and ac86u? Do I need to compile "dlarray_4366c0.bin" with a different setting? Thanks.

jlinktu, 2021-02-15:
I guess that the dhd module does not handle the `firmware_path` parameter.

The `dlarray_4366c0.bin` ends up on the wifi chip (ARM Cortex A7), which is the same for AC88U and AC86U, no need to compile it in a different way.

Can you locate the dhd.ko file on the router and give me the output of `readelf -s dhd.ko | grep dlarray_4366c0 | awk '{print $3}'`? It tells the size of the embedded firmware in the kernel module. It might be possible to exchange it if it is big enough. Otherwise, the easiest is probably to port the patch to the firmware that is currently runnig on your router.

quantumhub, 2021-02-19:
Dear @jlinktu 

Thank for your reply. I attached a copy of .ko file here, for research purpose:
[dhd.ko.zip](https://github.com/seemoo-lab/nexmon_csi/files/6011619/dhd.ko.zip)

And the output is here:
$ readelf -s dhd.ko |grep dlarray_4366c0 |awk '{print $3}'
0x10ab59

maxedbeech, 2021-02-28:
Great stuff opening this thread @quantumhub! 

Really appreciate your help with this @jlinktu - would love to help support this testing in any way I can to see if we can get it working on a ac88u.

quantumhub, 2021-03-05:
Thanks! It is wonderful if ac88u could be supported. :)

maxedbeech, 2021-03-15:
Hey @jlinktu! Would really appreciate to get your thoughts on @quantumhub's output if you have a sec, though equally am sure you're super busy and although I'm rather a layman, very happy to help analyse the output or try to port the software if you might be able to provide a few pointers?

Thanks so much!!
Max

quantumhub, 2021-05-17:
Thanks for your help! Currently, issue #210 is more urgent. A controllable packet injection is needed. Also obtaining agc and lna status is important. @maxedbeech @jlinktu