seemoo-lab/nexmon_csi #180: Timestamps from 4x4 MIMO CSI

quantumhub, 2021-02-07:
Dear authors,

Thank you for your great work! 
I get 4x4 CSI waveform successfully from asus ac86u. However, I think I do not correctly understand timestamps from 4x4 MIMO CSI. Could someone help to explain this? Thanks.

In theory, all these 4x4=16 SISO 80MHz CSIs are from the same 802.11 packet (e.g. with packet sequence number 721). So, they are from the same LTF field, which is with the duration of at most 4us*4=16us. 

However, in measurement, they show very different timestamps (frame.header.ts_sec + frame.header.ts_usec*1e-6).
The largest time gap can be up to 12ms (which is even longer than a usual TxOP duration, e.g. 3ms).
 
Log printed by matlab script (it is monitoring ping traffic, so 1 packet per second):
PktSeq721, TxStream 0-> RxAnt0 :  1487034809s  386978us <-
PktSeq721, TxStream 0-> RxAnt1 :  1487034809s  390412us
PktSeq721, TxStream 0-> RxAnt2 :  1487034809s  393882us
PktSeq721, TxStream 0-> RxAnt3 :  1487034809s  397291us
PktSeq721, TxStream 1-> RxAnt0 :  1487034809s  387951us
PktSeq721, TxStream 1-> RxAnt1 :  1487034809s  391249us
PktSeq721, TxStream 1-> RxAnt2 :  1487034809s  394757us
PktSeq721, TxStream 1-> RxAnt3 :  1487034809s  398155us <- (12ms later)
PktSeq721, TxStream 2-> RxAnt0 :  1487034809s  388721us
PktSeq721, TxStream 2-> RxAnt1 :  1487034809s  392122us
PktSeq721, TxStream 2-> RxAnt2 :  1487034809s  395626us
PktSeq721, TxStream 2-> RxAnt3 :  1487034809s  399029us
PktSeq721, TxStream 3-> RxAnt0 :  1487034809s  389554us
PktSeq721, TxStream 3-> RxAnt1 :  1487034809s  393010us
PktSeq721, TxStream 3-> RxAnt2 :  1487034809s  396506us
PktSeq721, TxStream 3-> RxAnt3 :  1487034809s  399900us

idx =

   639

PktSeq735, TxStream 0-> RxAnt0 :  1487034810s  387516us
PktSeq735, TxStream 0-> RxAnt1 :  1487034810s  391017us
PktSeq735, TxStream 0-> RxAnt2 :  1487034810s  394425us
PktSeq735, TxStream 0-> RxAnt3 :  1487034810s  397920us
PktSeq735, TxStream 1-> RxAnt0 :  1487034810s  388382us
PktSeq735, TxStream 1-> RxAnt1 :  1487034810s  391890us
PktSeq735, TxStream 1-> RxAnt2 :  1487034810s  395297us
PktSeq735, TxStream 1-> RxAnt3 :  1487034810s  398798us
PktSeq735, TxStream 2-> RxAnt0 :  1487034810s  389256us
PktSeq735, TxStream 2-> RxAnt1 :  1487034810s  392792us
PktSeq735, TxStream 2-> RxAnt2 :  1487034810s  396174us
PktSeq735, TxStream 2-> RxAnt3 :  1487034810s  399667us
PktSeq735, TxStream 3-> RxAnt0 :  1487034810s  390144us
PktSeq735, TxStream 3-> RxAnt1 :  1487034810s  393669us
PktSeq735, TxStream 3-> RxAnt2 :  1487034810s  397032us
PktSeq735, TxStream 3-> RxAnt3 :  1487034810s  400533us

idx =

   655

PktSeq738, TxStream 0-> RxAnt0 :  1487034811s  387720us
PktSeq738, TxStream 0-> RxAnt1 :  1487034811s  391117us
PktSeq738, TxStream 0-> RxAnt2 :  1487034811s  394611us
PktSeq738, TxStream 0-> RxAnt3 :  1487034811s  398106us
PktSeq738, TxStream 1-> RxAnt0 :  1487034811s  388588us
PktSeq738, TxStream 1-> RxAnt1 :  1487034811s  391989us
PktSeq738, TxStream 1-> RxAnt2 :  1487034811s  395489us
PktSeq738, TxStream 1-> RxAnt3 :  1487034811s  398898us
PktSeq738, TxStream 2-> RxAnt0 :  1487034811s  389454us
PktSeq738, TxStream 2-> RxAnt1 :  1487034811s  392850us
PktSeq738, TxStream 2-> RxAnt2 :  1487034811s  396360us
PktSeq738, TxStream 2-> RxAnt3 :  1487034811s  399765us
PktSeq738, TxStream 3-> RxAnt0 :  1487034811s  390332us
PktSeq738, TxStream 3-> RxAnt1 :  1487034811s  393724us
PktSeq738, TxStream 3-> RxAnt2 :  1487034811s  397229us
PktSeq738, TxStream 3-> RxAnt3 :  1487034811s  400626us

idx =

   671

PktSeq764, TxStream 0-> RxAnt0 :  1487034812s  388749us
PktSeq764, TxStream 0-> RxAnt1 :  1487034812s  392130us
PktSeq764, TxStream 0-> RxAnt2 :  1487034812s  395637us
PktSeq764, TxStream 0-> RxAnt3 :  1487034812s  399035us
PktSeq764, TxStream 1-> RxAnt0 :  1487034812s  389549us
PktSeq764, TxStream 1-> RxAnt1 :  1487034812s  393040us
PktSeq764, TxStream 1-> RxAnt2 :  1487034812s  396514us
PktSeq764, TxStream 1-> RxAnt3 :  1487034812s  399908us
PktSeq764, TxStream 2-> RxAnt0 :  1487034812s  390398us
PktSeq764, TxStream 2-> RxAnt1 :  1487034812s  393893us
PktSeq764, TxStream 2-> RxAnt2 :  1487034812s  397301us
PktSeq764, TxStream 2-> RxAnt3 :  1487034812s  400797us
PktSeq764, TxStream 3-> RxAnt0 :  1487034812s  391257us
PktSeq764, TxStream 3-> RxAnt1 :  1487034812s  394774us
PktSeq764, TxStream 3-> RxAnt2 :  1487034812s  398172us
PktSeq764, TxStream 3-> RxAnt3 :  1487034812s  401656us

jlinktu, 2021-02-07:
I guess you are referring to `frame.header.ts_sec` and `frame.header.ts_usec` read from a pcap in `utils/matlab/readpcap.m`?
That timestamp is generated by `tcpdump` when capturing frames and does not represent the time when the packet related to the CSI was received at the antenna. As it is not possible to pass several UDPs containing CSI to the host at the same time and `tcpdump` can only process incoming frames one by one too, the generated capture timestamps can vary that much.

quantumhub, 2021-02-07:
Thank you for your reply, @jlinktu 

After checking the source code, in  nexmon_csi/src/csi_extractor.c, 
https://github.com/seemoo-lab/nexmon_csi/blob/ba99ce12a6a42d7e4ec75e6f8ace8f610ed2eb60/src/csi_extractor.c#L104

I notice two places may related to the PHY rx timestamp. But what it the difference between these two? Besides, is any of these time stamp good enough for time of flight measurement? Thanks.

1)
    uint16 RxTSFTime;                   /* 0x014 RxTSFTime time of first MAC symbol + M_PHY_PLCPRX_DLY */
2)
    uint16 AvbRxTimeL;                  /* AVB RX timestamp low16 */
    uint16 AvbRxTimeH;                  /* AVB RX timestamp high16 */

yujianyuanhaha, 2021-02-24:
@quantumhub you can easily get the timestep using `tshark` command tool of Wireshark.

ozawali, 2021-06-29:
@quantumhub  I want to knwo how to configure to 4*4 MIMO. can you share with me the detailed steps ? Thanks

haochenku, 2021-07-23:
same here. @quantumhub , could you share the steps. In my side, all other tx CSI are 0s, only the first tx csi works for beacons.