seemoo-lab/nexmon_csi #181: PDP problem

ENB213-xiaoh, 2021-02-08:
Hi everyone. I'm currently working on raspberry pi 4B. I tried to plot the power delay profile of the received CSI by IFFT it. But, the peak always shows at the end of the plot. It seems that there are always 3 subcarriers offset. So, I'm wondering if anyone else has the same problem. Thanks.

[image]
The peak is supposed to be at the first time index as the distance is less than 5 meters.

quantumhub, 2021-02-09:
There is no time sychronization in WiFi. So, in theory, the first tap in observed channel impulse response can be arbitrarily shifted within OFDM CP duration.

closed 2021-02-10 by ENB213-xiaoh

haochenku, 2021-07-09:
Hi, @quantumhub @ENB213-xiaoh ,  how do you calculated the PDP? Did you set the unused subcarriers as 0? Currently I can see the peak jumps randomly . is there any way to make the PDP stable? what do you mean by no time sync in WiFi? I thought the symbol is synced by the preambles, right?

ENB213-xiaoh, 2021-07-09:
Yes. You do have to set null subcarrier to 0. That was caused by symbol timing offset, I mean that is why you see the peak jumps randomly. There is no way for us to make it stable, as the nexmon samples CSI in monitor mode without sync.

[EXTERNAL EMAIL] DO NOT CLICK links or attachments unless you recognize the sender and know the content is safe.

haochenku, 2021-07-09:
@ENB213-xiaoh   Thank you for the reply. This sync means the fine time sync, right? To detect the packet, the rx has to be able to find the beginning of the frame.

ENB213-xiaoh, 2021-07-09:
If you are dedicated to PDP using nexmon. I would recommend the following two papers.

"Decimeter Ranging With Channel State Information" TWC.19
"Precise Power Delay Profiling with Commodity WiFi" Mobicom.15

You don't have to read them all, just focus on the part of the symbol timing offset or packet detection delay.