seemoo-lab/nexmon_csi #182: Getting EOPNOTSUPP when tring to capture CSI on Asus RT-AC86U

itaysaadia, 2021-02-14:
Hi, and thanks ahead for your time.

I am trying to receive CSI packets with your software, running on Asus RT-AC86U (`nexutil -leh2 -V -Ieth6` attached here:)

```vendorid 0x14e4
deviceid 0x43c5
radiorev 0x2103eb
chipnum 0xaa90
chiprev 0x4
chippackage 0x0
corerev 0x41
boardid 0x798
boardvendor 0x14e4
boardrev P102
driverrev 0xa0a7a14
ucoderev 0x46843fd
bus 0x0
phytype 0xb
phyrev 0x21
anarev 0x0
nvramrev 0x0
```
platform             | firmware                         | vendorid | deviceid | radiorev   | chipnum | chiprev | chippackage | corerev | boardid | boardvendor | boardrev | driverrev | ucoderev  | bus | phytype | phyrev | anarev | nvramrev
-------------------- | -------------------------------- | -------- | -------- | ---------- | ------- | ------- | ----------- | ------- | ------- | ----------- | -------- | --------- | --------- | --- | ------- | ------ | ------ | --------
unknown              | 10.10.122.20 (r683106) FWID 01-e05466f |   0x14e4 |   0x43c5 |   0x2103eb |  0xaa90 |     0x4 |         0x0 |    0x41 |   0x798 |      0x14e4 |     P102 | 0xa0a7a14 | 0x46843fd | 0x0 |     0xb |   0x21 |    0x0 |      0x0

with Asus Firmware Version: 3.0.0.4.384_32799

I was able to compile and load your kernel module, and download nexutil and tcpdump as well. 
After the environment setup, I was trying to capture CSI
I used this makecsiparams 
(I tried to use both this parameters, and both the example from README.me and a similar one with my phone's MAC address. all of them failed): 
```bash
./makecsiparams -e 1 -c 157/80 -C 1 -N 1  # m+IBEQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA==
```

but every time I ran nexutil, I got a similar result:
`__nex_driver_io: error ret=-1 errno=95`
I checked, and errno 95 is EOPNOTSUPP (or Operation not supported on transport endpoint), but how does that make sense? eth6 is a valid and working WiFi interface! 

Thanks ahead,
Itay Saadia

zeroby0, 2021-02-14:
Did you bring up the interface and turn the radio on? I don't have much experience with ac86u, but I suspect that is the reason. You have to do that every time you reboot.

If you believe this is not a configuration error and is a bug in nexmon, you can use my pre-compiled binaries for rt-ac86u and instructions here: https://github.com/nexmonster/nexmon_csi/discussions/2. And few others and I have verified that these binaries work in the fashion described there, so you might be able to isolate the error. You could also try the version for Asus merlin-wrt we verified here: https://github.com/nexmonster/nexmon_csi/discussions/7

itaysaadia, 2021-02-14:
Thanks a lot! using the precompiled binaries, along with your explanation made this work!
closing the issue

closed 2021-02-14 by itaysaadia

zeroby0, 2021-05-08:
For people searching this in the future:
This error is because the modified Nexmon firmware module has not been loaded into the kernel properly.

You should [reload the module and bring up the interfaces](https://github.com/nexmonster/nexmon_csi/discussions/2) every time after rebooting the router. Or you can [set that to be automatically done](https://github.com/nexmonster/nexmon_csi/discussions/2) when the router reboots.