seemoo-lab/nexmon_csi #183: Abnormal CSI amplitude

quantumhub, 2021-02-15:
The figure is 4x4 MIMO measurements by ac86u in static environment. Two consecutive CSIs are shown.
The orange curves for Tx1 to Rx2 and Rx3 (marked by red arrows) are abnormal. Their amplitudes are doubled. 
It is neither due to Tx power change nor due to Rx LNA/AGC change, since they will apply to a row (all Rxs) or a column (all Txs).
So, it seems that the encoded CSI sequence is multiplied by 2 somewhere in software.
Could anyone help to fix this issue? Thank you!
[image]

jlinktu, 2021-02-15:
Shouldn't be a result of the firmware patch as it only reads and forwards the CSI values for the AC86U without modifying them.
Can you upload the corresponding capture file?
Is this reproducible and does it only happen for these two core/stream pairs?

quantumhub, 2021-02-15:
Dear @jlinktu ,

Thank you for your help!
It is reproducible and happens in all measured sessions. 
It happens to many tx/rx pairs.
The attachment is the raw CSI files. (The only change is the replacing of source mac address: sed 's/\xA8\x5E\xXX\xXX/\xA8\x5E\x12\x34/g' myRAWcsi3.pcap > temp3.pcap)

[bcm_CSI.zip](https://github.com/seemoo-lab/nexmon_csi/files/5983361/bcm_CSI.zip)

quantumhub, 2021-02-17:
Dear @jlinktu 

Thank you for your help!
The raw CSI is forwarded to tcpdump file without any change, and with good quality.

The doubling is due to a floating interpretation of exponent, in matlab parser. With this, I fixed the problem.
shft = nbits - maxbit;
https://github.com/seemoo-lab/nexmon_csi/blob/ba99ce12a6a42d7e4ec75e6f8ace8f610ed2eb60/utils/matlab/unpack_float.c#L57

closed 2021-02-17 by quantumhub

jlinktu, 2021-02-17:
Thank you for reporting.
The `shft` value should indeed only be recalculated when `autoscale` is enabled.
You might want to use a static `shft` value instead.

yangtat, 2021-07-02:
@quantumhub Thank you for work.
I found that the amplitude of CSI I collected also suddenly multiplied by 2. How did you solve it?