seemoo-lab/nexmon_csi #186: can't run matlab file

yael1994, 2021-02-22:
I try to run the matlab file csireader.m with example.pcap and i run the command:
root@raspberrypi:/home/pi/nexmon/patches/bcm43455c0/7_45_189/nexmon_csi/utils/matlab#./csireader.m
Usage: clear [options]

Options:
  -T TERM     use this instead of $TERM
  -V          print curses-version
  -x          do not try to clear scrollback
./csireader.m: line 2: fg: no job control
./csireader.m: line 3: fg: no job control
./csireader.m: line 4: syntax error near unexpected token `('
./csireader.m: line 4: `% read and plot CSI from UDPs created using the nexmon CSI extractor (nexmon.org/csi)'

what can i do?

Gi-z, 2021-02-22:
That script is a MATLAB script designed to be run in either [MATLAB](https://www.mathworks.com/products/matlab.html) or [GNU Octave](https://www.gnu.org/software/octave/index).

yael1994, 2021-02-22:
thank you very much!

closed 2021-02-22 by yael1994