seemoo-lab/nexmon_csi #189: Is my hardware connection setting correct to gather CSI?

dev-sora, 2021-02-25:
Hi,

I successfully installed the tool in ASUS RT-AC86U, and now wonder how the RT-AC86U work as CSI collector.  CSI is the information about wireless communication, so it's really strange for me that the RT-AC86U can listen the information even though its wireless capability is disabled while listening.  I also wonder what the interface eth6 do in CSI gathering.
I control the RT-AC86U via Ethernet cable by MacBook, and the WAN port of RT-AC86U connects with LAN port of another AP via Ethernet cable.  MacBook sends UDP packets to iPhone using iperf.  Both MacBook and iPhone communicate with the AP.  In my environment, this connection setting is the only way to listen CSI frames.  Is this setting appropriate?

quantumhub, 2021-02-26:
As long as there are wifi packet over air, the wireless channel frequency response between transmitter and your ac86u (monitor) can be measured. It doesn't need ac86u to participate communication.
You are using the right setup, since your AP to iphone link is over air.

dev-sora, 2021-02-26:
Hey @quantumhub , I'm really appreciate your kind reply!
Here is another question.  You said "the wireless channel frequency response between transmitter and your ac86u (monitor) can be measured", and does it mean the AP and the RT-AC86U also do communication?

quantumhub, 2021-03-01:
Since WiFi protocol does not have scheduling, all devices have to decode all traffic over air, at least including the destination  MAC address field, to avoid missing their target packets. AP does not need to communicate with RT AC86u.

dev-sora, 2021-03-05:
OK....
Then what they call CSI is the CSI shared between iPhone and the AP in my case? Or CSI which the ASUS router estimates from the frame communicating between iPhone and the AP?

dev-sora, 2021-03-05:
I assume that:
If a transmitter with 3 antennas and a receiver with 3 antennas communicate, the theoretical CSI matrix is a 3x3 matrix.  How can I obtain the matrix?
ASUS router has four antennas, and the parameters of makecsiparams is for the snififng ASUS router for core and spatial streams...  But I want to obtain the CSI ACTUALLY shared between the device(iPhone in my case) and AP.  How can I do that?