seemoo-lab/nexmon_csi #195: maximum package rate in ASUS router

yujianyuanhaha, 2021-03-15:
what is the maximum CSI package rate we can get in ASUS router? I am working on some activity monitor application, as comparison, the intel 5300 and AtherOS can go up to 1000Hz to 2500Hz.

I know we can set the rate by using different uCode from `makecsiprams`, and default is 40Hz (~600 package per second for 16 spatial stream), but higher rate may lead the router crash and reboot.

zeroby0, 2021-03-15:
Just a wild guess, but you may be able to get stability higher rates if you forward the traffic via ethernet to a computer, and then save it to pcap on that computer. You know, just incase the processor in the router is the bottleneck.

This is most probably wrong, but I'm basing it on the observation that tcpdump only collects about 300 packets per second while dumping to pcap, but shows more than 1000 packets per second when not dumping. So I'm guessing it's more processing intensive, so some packets are dropped. Although, the cli should print the number of packets dropped, and it somehow always prints 0 packets dropped.

yujianyuanhaha, 2021-03-15:
@zeroby0 I agree with your sense, the AtherOS tool note that running on router cannot get as high as that on laptop, and they are dump into `.bin` files at fixed rate.

zeroby0, 2021-03-16:
Ah, good to know I wasn't completely off the rails :) 

I wanted to search information about whether tcpdump drops and packets, but couldn't find any. Cloudflare has XDPcap which I think captures packets before they go through the kernel's network stack (I'm not completely sure), and might be able to capture more packets than tcpdump.

quantumhub, 2021-03-16:
@matthiasseemoo @zeroby0 @yujianyuanhaha 
Since the UPD is with destination address 255.255.255.255, it is supposed to be broadcasted in LAN automatically, and minimize CPU load on ac86u:
[prepend_ethernet_ipv4_udp_header(p_csi)](https://github.com/seemoo-lab/nexmon_csi/blob/ba99ce12a6a42d7e4ec75e6f8ace8f610ed2eb60/src/csi_extractor.c#L251);

So, in theory, tcpdump on a PC within same LAN is expected to receive UDPs containing CSI from port 5500. 
But, nothing is received at the PC, in my measurement. Please let me know any of my understanding is wrong. Thank you very much!

zeroby0, 2021-03-16:
Probably because they are in-bound packets. I think only outbound packets to 255.255.255.255 are sent out of your device to the router to be broadcast.

quantumhub, 2021-03-16:
I also tried unicast address:

The broadcast address below is replaced by an unicast one:
[.dst_ip.array = { 255, 255, 255, 255 }](https://github.com/seemoo-lab/nexmon/blob/57a2fb0c3e4d774f32c96160dab9048b888a1d28/patches/common/udptunnel.c#L66)
to 
.dst_ip.array = { 10,10,10,123 }

The change seems to be effective, and tcpdump captures packet to 10.10.10.123 from 10.10.10.10.
However, my PC with IP address 10,10,10,123 can't get anything from port 5500.
(I also use ifconfig to assign IP 10.10.10.10 to ac86u eth6. And the IP is reachable by PING (between PC and ac86u). It confused me that it still does not work.)

zeroby0, 2021-03-16:
Yes, cause they are **inbound** packets.

As described in the paper, the firmware in the chip adds these new headers and sends it to the kernel. The kernel sends it to tcpdump. No where in this process are the packets going **out** of the CSI device. Even though the incoming packets are addressed to 10.10.10.123, the kernel will not redirect them to the correct IP because they are **Incoming** packets.

If you create a packet from userspace, and send it to kernel, and then kernel sends it to the firmware in the WiFi/Ethernet port. Then the WiFi/Ethernet port sends it to the Router, which redirects it to your PC. These are **outbound** packets.

To collect packets on your laptop, you need to take the packets *incoming from wlan0* (in RPi), and then send them *out of the device* via Eth0.

You can either do this via IPtables (difficult, but likely more performant), or via [SSH port forwarding (trivially easy)](https://linuxize.com/post/how-to-setup-ssh-tunneling/#remote-port-forwarding), or maybe even via a python script which listens to wlan0 port 5500 and puts packets received from it into Eth0 to your laptop's ip.

quantumhub, 2021-03-16:
Thank you for providing these solutions!
The easy solution seems to be interesting.

I tried to run it on ac86u:
Method 1:
ssh -L 5500:my-laptop-IP:5500 username@my-laptop-IP
Method 2:
ssh -R 5500:127.0.0.1:5500 -N -f username@my-laptop-IP

Also, I tried on my laptop, but get error message.
Method 3:
ssh -R 5500:localhost:5500 username@ac86u-IP
Warning: remote port forwarding failed for listen port 5500
Probably I used wrong ssh configuration...

Could you show which ssh port forward command I should use? Thank you!

zeroby0, 2021-04-11:
@quantumhub The radio is turned off when sending CSI from chip to kernel, so if a new packet arrives in the air when the radio is off, it will be lost. Maybe this is why your MAC sequences are not continuous.

Did the rate increase when you forwarded collection to your PC? Using IPTables might do the packet forwarding in the kernel itself, which might make it more performant.

zeroby0, 2022-11-13:
I found a way to forward packets using iptables/nftables. Port forwarding via SSH works too, but it's not as efficient because of the encryption & decryption. Although I doubt the efficiency matter so much as were only sending a few thousand packets per second, I'm gonna leave a link for people of the future to find: https://github.com/nexmonster/relay