seemoo-lab/nexmon_csi #196: MUSIC for ToF

stepanmazokha, 2021-03-25:
Hi all,

In the paper "Free Your CSI: A Channel State Information Extraction Platform For Modern Wi-Fi Chipsets", there's a section where Nexmon authors evaluate CSI by testing ToF measurement using MUSIC algorithm (Section 3.2).

There are a few other publications that are trying to accomplish similar results, yet few have achieved similar accuracy. Hence my question is mostly directed toward the authors (unless someone in the community can confirm/correct my questions, of course!) 

So, here are my questions:

1. Are you relying on any CSI compensation techniques in your work? The Nexmon CSI extractor provides CSI after AGC has been applied. To the best of my knowledge, this causes CSI to lose most of the range-related information. 

2. Do you apply MUSIC directly to Nexmon CSI measurements? Do you somehow recover "raw" CSI using RSSI or any other techniques?

3. Finally, you mention "sweeping through channels [1..13] to increase the spatial resolution". Does this mean that you are "hopping" between all channels all the time? Wouldn't this make CSI inconsistent (because it's not all collected at the same moment in time across all subcarriers)? Is the channel change speed "fast enough" so that you could capture CSI of the same transmitted frame on all channels?

In advance, thank you!

yujianyuanhaha, 2021-07-01:
@stepanmazokha I feel like there are also questions on hardware side when apply MUSIC
(1) for ASUS router, which spatial stream corresponding to the external arrays? And how to deal with the antenna spacing, we expect less half-wavelength for ULA, it is about 9cm in ASUS router, larger than 6.25cm (half wavelength at 2.4G) and 3cm (half wavelength at 5G)  
(2) how to deal with the phase offset, "Tewes, Simon, and Aydin Sezgin. "WS-WiFi: Wired Synchronization for CSI Extraction on COTS-WiFi-Transceivers." IEEE Internet of Things Journal 8.11 (2021): 9099-9108.", it mentioned the temperature and attennas imperfection that bring the offset.

stepanmazokha, 2021-07-06:
@yujianyuanhaha, thanks for responding!

1.1. (which spatial stream corresponding to the external arrays): I determined this experimentally, and the response in the issue you created matches my findings. [this one](https://github.com/seemoo-lab/nexmon_csi/issues/199#issuecomment-824455183).

1.2. (And how to deal with the antenna spacing): probably need to figure out a way to connect these antennas to some external board with a configurable antenna distance, and wire them up with a proper antenna order.

2. Temperature-related differences are an interesting note, I haven't seen this work just yet, thank you!

Also, thanks for referring to the mD-Track!