seemoo-lab/nexmon_csi #198: Chip Goes Down After A While - Chanspec Output Seems Wrong

Adam-Antios, 2021-03-29:
I build and install the nexmon csi tool in a Raspberry Pi Model 3 B+ with RPi OS using kernel 5.4.83-v7+.

I check that the chanspec is correctly set and the command `nexutil -m` produces a 2, but after a minute or so the chip goes down as the chanspec produces the code `0x6863 85/160` and the `nexutil -m` gives a 0. This happens whether I run` tcpdump` and gather packets, or do absolutely nothing. 

Moreover, at any time if I try to set the channel and the bandwidth as 1/20 instead of getting as a result an output 1/20 I get 1 which seems wrong. I thought that this might have been a formatting issue and thus I tried setting it as 01/20 or 001/20, with no different results. In order to use this channel and bandwidth I changed the `src/regulations.c: additional_valid_chanspecs[]` first.

Asafe33, 2024-06-04:
were you able to find a solution?

Adam-Antios, 2024-06-05:
Unfortunately no, I am sorry.

Asafe33, 2024-06-05:
so which way did you go? How did you capture the csi data?

Adam-Antios, 2024-06-12:
I didn't. I abandoned the effort due to the experimental nature of the project. It wasn't mature enough for my purposes.

closed 2024-06-13 by jlinktu