seemoo-lab/nexmon_csi #199: Locate the 4 physical antennas in ASUS router

yujianyuanhaha, 2021-03-31:
As I aim to apply the spotFy  idea on nexmon CSI Phase, we need to apply the calibrate method first to ease the “external path error”

[image]

zeroby0, 2021-03-31:
Wikidevi might have FCC applications which have images with more information:

https://wikidevi.wi-cat.ru/ASUS_RT-AX86U
https://deviwiki.com/wiki/ASUS_RT-AC86U
https://fccid.io/document.php?id=3365365

OpenWRT and Merlin wiki's might have more info as well. Good luck :)

quantumhub, 2021-04-01:
Hi, @yujianyuanhaha 

By using a long RF cable to extend one of the antenna to a remote place, and disturb the channel for this specific antenna, you will observe a row or a column in MIMO channel matrices fluctuates significantly. Then, you can identify its antenna port index.

Or let one of the port's signal pass through an attenuation (e.g. 20dB), you can also find it by per Rx antenna RSSI.

You don't have to tear it down. :)

Cyrbaby, 2021-04-07:
you can use the power splitter

quantumhub, 2021-04-21:
This figure shows the indices for ac86u's antennas:
[image]

Cyrbaby, 2021-04-22:
> This figure shows the indices for ac86u's antennas:
> [image]

Thanks for your reply， may I ask if I use the ac86u as the transmitter， whether the index of spatial stream is the same as the index in the figure？

quantumhub, 2021-04-22:
Hi, @Cyrbaby 
If your transmission is with 4 spatial stream, there should be one to one correspondence between stream id and the antenna port id shown above.
However, since we don't have control of the MCS of transmitted frames, the stream number <=3 may happen. In the case of stream number <=3, each stream is spread over all antennas by some spatial precoding vector (determined by driver or ucode inside).

Cyrbaby, 2021-04-23:
@quantumhub thank you for your answer ：）

yujianyuanhaha, 2021-06-15:
@quantumhub  thanks.
Do the 2.4G signals only go by the external antennas, while 5G signal only go by the inner antennas (the square-shape antenna)? What is the specific spatial stream index meaning, does 1st spatial stream mean RX1-TX1, while 2nd for RX2-TX1 ...?

[image]

When I do data collection on 5G, the CSI spatial stream histogram is like
[image]

While for 2.4G it is like 
[image]

RenjieZhao, 2021-07-06:
Hi @yujianyuanhaha 

1. We also tear down the router to see the exact connection. The internal antenna is only for 5GHz and external antennas are for 5GHz and 2.4GHz. The chipset has separate pins for 2.4GHz and 5GHz bands. The internal antenna only connects with one set of pins so it should only work for one frequency band. For the external antennas, there are diplexers to combine the 2.4GHz and 5GHz so that they can share the same antenna. You can see this if you remove the shield. This is the reason why you see that the result of antenna 2 at 2.4GHz is always 0.

2. I guess so. Your result matches @quantumhub 's antenna index.

One thing which might cause an issue is that the 4 by 4 MIMO in the router is implemented by two 2 by 2 chips. The two chips might only shares reference clock instead of LO. This means that the frequency of the carrier on the two chips are the same, but the phase might change. Calibration might be needed after every booting. Did you see this issue?

yujianyuanhaha, 2021-07-07:
@RenjieZhao thanks for advice. I never look into the two chip issue and their phase different.

munir01, 2021-09-09:
Thank you for investigating this. This is very helpful. Does anyone know which antennas are connected to which Wi-Fi chipset? Is it the case that by following the antenna numbers of @quantumhub, antenna 0 and 1 are connected to chipset 1 and antenna 2 and 3 are connected to chipset 2? Thank you!