seemoo-lab/nexmon_csi #201: Cannot connect to WiFi and Cannot get CSI

albert0527, 2021-04-07:
Dear all, I use this tool on Raspi 4B , the version is kernel 5.4.83-v7l+

the wifi is 5GHz(154 channel) and 2.4GHz(7 channel and 20 channel) 

When I enter the created subdirectory nexmon_csi and run "make install-firmware"(Step 10), the WiFi always lost its connection. That means I cannot visit the website. Actually, in the router management I can see it is connected to the router. Is it normal? And for checking the Internet access, I cannot ping my default gateway successfully.

So, I cannot get any CSI data and the cap documents (xx.cap) only indicate 4096 bytes 4 Kb or 0 kb many times. If I try to open these documents using whireshark, they are all empty.
Sometimes, these caps will be full of APR protocols (about 10). So I thought it could be the Internet connection problem.

Here is  the whole process:
1. Since the kernel have not support 5.10 right now (if update the Raspberry the kernel will update to 5.10), I did not doing "apt-get update && apt-get upgrade" during the whole progress.

2.sudo apt-get install tcpdump texinfo automake autoconf libtool 

3.  sudo su
     apt install raspberrypi-kernel-headers git libgmp3-dev gawk qpdf bison flex make
     git clone https://github.com/seemoo-lab/nexmon.git
       cd nexmon
                cd buildtools/isl-0.10, ./configure, make, make install, ln -s /usr/local/lib/libisl.so /usr/lib/arm-linux-gnueabihf/libisl.so.10
                 cd buildtools/mpfr-3.1.4, autoreconf -f -i, ./configure, make, make install, ln -s /usr/local/lib/libmpfr.so /usr/lib/arm-linux-gnueabihf/libmpfr.so.4
       cd nexmon
               source setup_env.sh
               make
       cd patches/bcm43455c0/7_45_189/
               git clone https://github.com/seemoo-lab/nexmon_csi.git
       cd /home/pi/nexmon/patches/bcm43455c0/7_45_189/nexmon_csi
              make install-firmware
       cd /home/pi/nexmon/utilities/nexutil/
            make && make install

4.                        Extraction
     cd /home/pi/nexmon/patches/bcm43455c0/7_45_189/nexmon_csi/utils/makecsiparams
     make
     iw dev
     iwlist wlan0 scan | grep -C 5 "XXXXX"(Wi-Fi name)
     ./makecsiparams -c 11/20 -C 1 -N 1 -m A8:BD:27:E1:36:02
     ifconfig wlan0 up
     nexutil -Iwlan0 -s500 -b -l34 -vm+CxABEQAAAQCovSfhNgIAAAAAAAAAAAAAAAAAAAAAAAAAAA==
     iw phy `iw dev wlan0 info | gawk '/wiphy/ {printf "phy" $2}'` interface add mon0 type monitor
     ifconfig mon0 up
     tcpdump -i wlan0 dst port 5500 -w ./1001.cap

albert0527, 2021-04-07:
[image]
[image]

zeroby0, 2021-04-08:
Hi!

Please skim through the previous issues in the repo, they answer why you can't connect to WiFi, and also show you how to solve 99% of the problems that arise when using Nexmon.

albert0527, 2021-04-08:
> Hi!
> 
> Please skim through the previous issues in the repo, they answer why you can't connect to WiFi, and also show you how to solve 99% of the problems that arise when using Nexmon.

sorry, I tried the methods mentioned in the previous issues, But I still cannot connect to the internet and capture packets. Maybe it is a new promblem.

Gi-z, 2021-04-08:
> sorry, I tried the methods mentioned in the previous issues, But I still cannot connect to the internet and capture packets. Maybe it is a new promblem.

I'm sure I'll be corrected if this is wrong, but my understanding is that while using `nexmon_csi` in CSI extraction mode your NIC is operating is monitor mode. This means your device is unable to associate with other wireless devices operating in managed mode (e.g. normal WiFi). If you need to use your device over SSH, I would suggest using Ethernet, as you are unable to simultaneously use `nexmon_csi` in CSI extraction mode and communicate with the device over WiFi.

albert0527, 2021-04-12:
> I'm sure I'll be corrected if this is wrong, but my understanding is that while using `nexmon_csi` in CSI extraction mode your NIC is operating is monitor mode. This means your device is unable to [...]

sorry, maybe I did not express my meaning very clearly. Actually, I found after installing firmware, I cannot ping my routers successfully and on the router management panel  it did not show the Pi's connection. But at this time, I can find it was connected to the router in the WiFi panel in the upper right corner of the screen. If use the command 'ifconfig', I would find the IP address will change to '169.254.199.147'. Maybe this version firmware has affected the DHCP service so the PI cannot get the correct IP address and use the 'Link-local address'? Even I used the static IP address, I cannot Ping my router on the Pi successfully. 
So maybe I think this failed connection between Pi and the router influenced the CSI capture. And I cannot get CSI  data, so the PI was always broadcasting ARP to find correct IP address for connection.
Emmm, I didn't find similar question in the previous issue. I don't know whether others have encountered this problem. 
Thank you very much!

dejavo4e, 2021-04-27:
could u solve your problem?@albert0527

marioqwert9, 2021-06-02:
I have the same problem, wlan0 is deleted so i dont know how to connect do wi fi and collect csi packet

Saberfish, 2021-09-10:
Me too! The wlan0 is deleted and I don't now how to load the firmware sice I reboot the pi.

sangeun-jo, 2023-06-02:
It is totally normal. You don't need wifi to capture CSI. 
Try other channel and bandwidth.

Asafe33, 2024-06-05:
@sangeun-jo  @marioqwert9  @Saberfish @albert0527 Could you explain better? I still don't fully understand.