seemoo-lab/nexmon_csi #202: "Recipe for target 'gen/ucode.bin' failed on Asus RT-AC86U router installation error

GerbenGheysens, 2021-04-13:
[image]
When I want to install the firmware on the Asus RT-AC86U router I've got an error message about the gen/ucode.bin target. 
Does anyone know how to solve this issue?

zeroby0, 2021-04-13:
Did you source the setup file before building?

I also have pre-compiled binaries here that you can use: https://github.com/nexmonster/nexmon_csi/discussions/2

GerbenGheysens, 2021-04-13:
Yes, setup was done properly, but I will try with the pre-compiled binaries :) thx!

zeroby0, 2021-04-13:
There should be compiler logs somewhere nearby in the directory structure, they might contain a more verbose error message.

The pre-compiled binaries will reduce the work you need to do, but if you need to build it for some reason, I have the build guide here: https://github.com/nexmonster/nexmon_csi/discussions/7.

closed 2021-04-13 by GerbenGheysens