seemoo-lab/nexmon_csi #204: CSI packets are captured but pcap file is empty

dejavo4e, 2021-04-18:
hi;
I tried to capture CSI packets using "tcpdump -i wlan0 dst port 5500 -vv -w output.pcap -c 1000"
the result shows that 1000 CSI packets were captured.
I mount my usb flash drive and "cp" the pcap file. But the files are either 0kb or 325kb.
I didn't find similar problem . I don't know whether others have encountered this problem or not.
Thank you for your respond.

parisafm, 2021-04-19:
Hi there; Try to reinstall raspian and nexmon. Be sure that you are using the right kernel version. Check the version of kernel header.
It should be 5.4.  U can check its version by this command:
uname -r

By the way, I recommend u to do as it is said in https://github.com/zeroby0/nexmon_csi/tree/pi-4.19.97

dejavo4e, 2021-04-29:
@parisafm 
Thanks for your respond. I have checked my kernel by using "uname -a" . it is 5.4.79. 
still i get empty pcap files. 
here is my results:
tcpdump: listening on wlan0, link-type EN10MB (Ethernet), capture size 262144 bytes                                                                         
1000 packets captured
1120 packets received by filter
0 packets dropped by kernel
root@raspberrypi:/home/pi# du -h a20.pcap
328K    a20.pcap

yujianyuanhaha, 2021-05-01:
@dejavo4e  what about longer time capture, mine is like `./tcpdump -i eth6 dst port 5500 -vv -G 5 -W 1 -w /mnt/MYDISK/test.pcap;` where `-G 5` mean 5 seconds collection.

Best

zeroby0, 2021-05-08:
Are you sure the pcap file is empty?

If your WiFi bandwidth is 20 MHz, the ~300KB file size seems reasonable. An 80 MHz file with 1000 samples is about 1 MB.

You can use [CSI-Explorer](https://github.com/nexmonster/nexmon_csi/tree/feature/python/utils/python) to inspect your CSI files and see whether they're really empty. Or you can open them up in Wireshark.

dejavo4e, 2021-05-08:
> 
> 
> @dejavo4e what about longer time capture, mine is like `./tcpdump -i eth6 dst port 5500 -vv -G 5 -W 1 -w /mnt/MYDISK/test.pcap;` where `-G 5` mean 5 seconds collection.
> 
> Best

same as before. pcap file is empty

dejavo4e, 2021-05-08:
> Are you sure the pcap file is empty?
>
> If your WiFi bandwidth is 20 MHz, the ~300KB file size seems reasonable. An 80 MHz file with 1000 samples is about 1 MB.
>
> You can use [...]

yeah i tried wireshark. still pcap file is empty. I reinstalled raspian but it didn't work as expected...

zeroby0, 2021-05-08:
Just to confirm, the 325 kb file had no data? It might be a problem with tcpdump or your setup.  Try using channel 36, and make sure there is wireless traffic in that channel when you collect.

I'm using kernel 5.4.51, I created a fresh install of nexmon_csi and it works as expected. https://github.com/nexmonster/nexmon_csi/tree/pi-5.4.51-plus

yjxb1, 2023-08-04:
Hello, have you solved the problem? The bag I caught is also empty。