seemoo-lab/nexmon_csi #206: Can't really understand the amplitude of the CSI read from official example.pcap file

Colin-Guolin, 2021-04-19:
Hi everyone I used the official Matlab program to read the [example](https://github.com/seemoo-lab/nexmon_csi/blob/master/utils/matlab/example.pcap)`.pcap `file. But the amplitude of the CSI doesn't make sense, 
The channel statement information in 802.11 is calculated by H = Y/X, where H is the CSI, Y and X are the transmitted and the received long training sample in the frequency domain. The value of this should have ranged from 0 to 1, 0 means the signal faded significantly and 1 means the signal has no fading after the channel. Due to the multipath channel, the value could be a little bit larger than 1. However, the amplitude of the plot ranged from 0 to 30000. Could anyone explain this?
[image]

Thanks in advance

quantumhub, 2021-04-22:
There must be something wrong in your data reading or plot. The amplitude curve should be very smooth. The extracted CSI has already been processed with VHT-LTF pilot sequence. So, the readouts are coefficient in H. They are scaled by some multiplier factor(analog and digital filtering), so their amplitudes does not need to be within (0,1).

yangtzhs, 2021-04-24:
Hi，I am doing this project now, and using example .pcap by Matlab, then get the correct 
data, but when I update the chip version bcm4339, I got the wrong data analysis, which is simlar to yours, so I advise you to check the chip vision, hope it will help you .
Looking forward to your reply.
[image]
[image]

haochenku, 2021-07-07:
Hi @yangtzhs , what do you mean by update the chip 4339 version? does that mean bcm4339 has different versions?
what I found is that data csi is much lower than the beacon csi. Does the following look good? 
[image]