seemoo-lab/nexmon_csi #207: Support for CSI extraction for latest intel Wi-Fi is available in Linux kernel

atlury, 2021-04-24:
Dear Friends,

Thank you all for the fantastic work that you guys are doing with Nexmon-csi. I will like to bring to your attention the following interesting CSI support in intel Linux wireless drivers. If you get time please do look into it and integrate them.

[08/21] iwlwifi: mvm: implement CSI reporting

> `https://patchwork.kernel.org/project/linux-wireless/patch/20190127101908.6428-9-luca@coelho.fi/`

IWL_MVM_VENDOR_CMD_CSI_EVENT: CSI event, use as a command to enable unicast reporting to the calling socket

> `https://git.kernel.org/pub/scm/linux/kernel/git/iwlwifi/backport-iwlwifi.git/tree/drivers/net/wireless/intel/iwlwifi/iwl-vendor-cmd.h?h=release/core59`

struct iwl_csi_chunk_notification

> `https://git.kernel.org/pub/scm/linux/kernel/git/iwlwifi/backport-iwlwifi.git/tree/drivers/net/wireless/intel/iwlwifi/fw/api/location.h?h=release/core59`

There are two papers published that actually use the latest Intel Wi-Fi chips for CSI

1. The "On Phase Offsets of 802.11ac Commodity WiFi - Anatolij Zubow, Piotr Gawlowicz, Falko Dressler" uses the modern commodity 802.11ac chips like Intel 9260. More precisely they backported the release/core46 to Linux 5.5.1 kernel. 

* Apparently the authors ran it in monitoring mode (supported on most intel Wi-Fi cards as per https://wireless.wiki.kernel.org/en/users/drivers/iwlwifi )

* About the monitor / sniffer mode as per https://wireless.wiki.kernel.org/en/users/drivers/iwlwifi
Intel says their devices support monitor mode. When you have VHT APs around, you should load the iwlwifi module with `amsdu_size=3` This will put lots of pressure on the memory subsystem, but it will allow you to hear 12K long packets. 

* Each packet was captured probably using say the management frame (CSI is in the management frame) got through nl80211.h from userpace via libnl Netlink API.
https://git.kernel.org/pub/scm/linux/kernel/git/iwlwifi/backport-iwlwifi.git/tree/include/uapi/linux/nl80211.h  

* It is known that as per https://wireless.wiki.kernel.org/en/developers/p2p/overview NL80211_CMD_REGISTER_FRAME - Allows a user space application to register for receiving a given type of (management) frame through nl80211, and also replying to it. Applications can also specify a filter so for example they don't have to handle all the different action frames but just a subset. Further, currently only one process can register to receive these frames as per https://www.spinics.net/lists/linux-wireless/msg129677.html

* The authors in the above paper said, they reverse engineered the intel specific encoded CSI frame by comparing results with old intel 5300 NIC.

2. The "Sensor-Aided Learning for Wi-Fi Position With Beacon Channel State Information J Choi Intel" paper uses the latest AX200 series Wi-Fi chip apparently with a custom tool from intel themselves.

I hope this will benefit the community a lot if someone who has time can take it up and integrate it with Nexmon.

atlury, 2021-04-24:
Further to the above

There is a iw: add command to register and dump mgmt frames below
> https://patchwork.kernel.org/project/linux-wireless/patch/20171118182052.15515-1-sergey.matyukevich.os@quantenna.com/

> https://patchwork.kernel.org/project/linux-wireless/patch/20171014210036.29556-1-sergey.matyukevich.os@quantenna.com/

atlury, 2021-04-24:
Adding more information. As per the discussion here https://www.spinics.net/lists/linux-wireless/msg167153.html

"The CSI data has variable length mostly depending on the bandwidth but it's fundamentally always tied to a specific frame and as such we've always attached it to that frame using a radiotap vendor namespace.

You can easily implement that in a mac80211 driver since it has support for that via RX_FLAG_RADIOTAP_VENDOR_DATA and the associated struct ieee80211_vendor_radiotap that you put into the skb's head"

quantumhub, 2021-04-29:
Interesting to see this news!
Thanks for sharing! @atlury

jlinktu, 2021-05-05:
thank you for the detailed listing, @atlury 

however, I'm not sure what you'd like to integrate with nexmon here
the nexmon_csi patch forwards CSI to the host via UPD frames, a generic solution that allows receiving CSI by any application that can listen to a socket without any further driver modifications - why would you want to introduce additional Management frames with Radiotap headers and involve netlink?

atlury, 2021-05-08:
@jlinktu 

Some more observations

Intel allows csi (channel state information) reporting as per which used to be attached per frame.
[08/21] iwlwifi: mvm: implement CSI reporting
https://patchwork.kernel.org/project/linux-wireless/patch/20190127101908.6428-9-luca@coelho.fi/

However it was moved out of band (no longer tied to any packet/frame so things like radiotap etc wont work) as per below
https://git.kernel.org/pub/scm/linux/kernel/git/iwlwifi/backport-iwlwifi.git/commit/?id=3d380b6b301f504c8f3aa2f64d1cef60821acb77

However MVM firmware vendor-cmd.c still captures csi (apparently without attaching it to any frame)
https://github.com/intel/backport-iwlwifi/blob/e3a6af8dfaebc0be3d05099459d5a643bd438931/iwlwifi-stack-dev/drivers/net/wireless/intel/iwlwifi/mvm/vendor-cmd.c#L1887

So yes probably you are right, both are completely different and hence not as straightforward. Just posting this info if its useful for others. I will take up the work sometime when I get free.

Thanks

jiangzp, 2021-10-09:
PicoScenes CSI platform now supports the MVM-based Intel Wi-Fi 6AX200 NIC, which can report 1992x2x2 CSI values in the 160 MHz 2x2 MIMO case. For more information, you may refer to [ps.zpj.io](ps.zpj.io).

atlury, 2021-10-09:
Congrats! It should now be easily extendable to other series from Intel. Nexmon could also benefit from this work.

jiangzp, 2021-10-09:
Yes, it can also extend to AC9260 or even AC7260 models. However, these models are relatively old and only with 11ac support.

 So, I focus on developing CSI support for AX200 and the upcoming AX210.

atlury, 2021-10-10:
I am happy to help but yeaa we can leave it to Nexmon to see how they can do it as they better understand it and let me try out the picoscenes integration work. I will write to you.