seemoo-lab/nexmon_csi #208: 0 packet captured and Empty File and Can't capture CSI PI4

serrhini, 2021-05-04:
Thank you for your work, after installing nexmonCSI on rasp Pi4 , kernel 4.19.97-v7l+, I started the CSI packets capture but pcap file is empty and no packet was received, please can you enlight me as I m new tester.
plz see my console bellow

pi@raspberrypi:~ $ uname -r
4.19.97-v7l+
root@raspberrypi:/home/pi# mcp -c 157/80 -C 1 -N 1 m+IBEQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA==
m+IBEQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA==
root@raspberrypi:/home/pi# ifconfig wlan0 up
root@raspberrypi:/home/pi# nexutil -Iwlan0 -s500 -b -l34 -vm+IBEQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA==
root@raspberrypi:/home/pi# iw dev wlan0 interface add mon0 type monitor
root@raspberrypi:/home/pi# ip link set mon0 up
root@raspberrypi:/home/pi# tcpdump -i wlan0 dst port 5500 -vv -w output.pcap -c 1000
tcpdump: listening on wlan0, link-type EN10MB (Ethernet), capture size 262144 bytes
Got 0
^C0 packets captured
0 packets received by filter
0 packets dropped by kernel
root@raspberrypi:/home/pi#

zeroby0, 2021-05-08:
Hey @serrhini ,

Try using channel 36. The higher channels seem to be buggy on a lot of Broadcomm chips. You will have to configure your WiFi router to Channel 36, and connect another WiFi device to that router and create some wireless traffic. Then use -c 36/80 and you should be able to collect CSI.

serrhini, 2021-05-12:
Hi thank, I apreciate lot , I succefully collected CSI.

closed 2021-05-12 by serrhini

kinyip3, 2025-03-03:
hi @zeroby0 
I have the same issues on Pi4B with 5.10.92-v7l+ capturing 36/20 
tcpdump: listening on wlan0, link-type EN10MB (Ethernet), capture size 262144 bytes
Got 0
^C0 packets captured
0 packets received by filter
0 packets dropped by kernel
what should i check?

i capture 36/20 as i check iw dev 
the interface wlan0 shows channel 36(5180 MHz), width: 20Mhz, center1: 5180 MHz, is this correct?