seemoo-lab/nexmon_csi #209: whether all 4 antennas of ac86u router are transmitting data all the time？

zel602, 2021-05-05:
Thanks for sharing the great project！
I have a question want to ask，assuming a 2 antennas device is connected to a 4 antenna router, for example , ac86u. Only 2× 2MIMO can be used between the router and the device. Can I assume that only two antennas at the router end are transmitting data, there are two spatial streams, and the remaining two antennas are not working and not transmitting data?

Thanks in advance for reply.

matthiasseemoo, 2021-05-05:
No necessarily, you can transmit two streams over four antennas and receive them with two antennas.

zel602, 2021-05-05:
@matthiasseemoo thank you for the answer! About relationship between stream and antenna, could you recommend some articles or websites for studying?

quantumhub, 2021-05-07:
You can find it on wiki:
https://en.wikipedia.org/wiki/Precoding#Description_of_Multi-user_MIMO

x=sum{ws}, where w is the precoding vector, which spread a spatial stream to all active antennas.

Cyrbaby, 2021-05-08:
@quantumhub thank you for your help！I understand a spatial stream may be transmitted from more than one antenna.

zel602, 2021-05-08:
@quantumhub thank you for your help! I understand a spatial stream will be transmitted from more than one antennas