seemoo-lab/nexmon_csi #210: Can the Asus ac86u be used as transmitter to measure angle of departure？

zel602, 2021-05-09:
Thanks to everyone's help，I understand a spatial stream will be transmitted from more than one antennas.

May I ask anonther question, Can I use the Asus ac86u as transmitter to measure angle of departure？

I know the premise of the question is that the positional relative relation of the spatial streams is known,  is it possible to obtain this prior knowledge?

yujianyuanhaha, 2021-05-15:
@zel602 It is possible to measure the angles between a pair of transmitter and receiver router using [SpotFy](https://web.stanford.edu/~skatti/pubs/sigcomm15-spotfi.pdf).

zel602, 2021-05-17:
@yujianyuanhaha thanks for your answer！But I care about how to capture the valid data to obtain angle of departure, rather than the theory.

quantumhub, 2021-05-17:
More effort is needed to make this tool useful. 
The agc and lna status associated with CSI have to be extracted (but not yet provided by this tool). Otherwise, it can't be used for aoa. For aod, preceding matrix has to be fixed. But now, it is in wild status. @yujianyuanhaha @zel602

zel602, 2021-05-18:
@quantumhub  Thank you for your answer！ It is really helpful！

yujianyuanhaha, 2021-05-19:
@quantumhub I assume AoA can be estimated using the [SpotFy](https://web.stanford.edu/~skatti/pubs/sigcomm15-spotfi.pdf). AGC and LNA can be skipped.

quantumhub, 2021-05-19:
Spotfi is based on Intel 5300 NIC card. I have not tried this card. 
Intel 5300's CSI extraction ucode implementation is proprietorial, unlike Atheros ath9k CSI or Nexmon Bcm here. 
Maybe Intel's proprietorial implementation has compensate lna and agc's impact. Or they have fixed lna and agc's gain level. @yujianyuanhaha