seemoo-lab/nexmon_csi #211: how to connect the router and the xubuntu PC？

wolke1526, 2021-05-13:
Hi everyone, I'm trying to do it with a Asus ac86u router and I have a few questions:
1. how should I connect the router and the PC If I want write this firmware on the router? can they be connected with a LAN cable?
2. Should I use another ac86u router as an AP to transmit the signal? or any other 11ac/n device can do the work?
I'm new to this project but very interested in it, will be really appreciated if someone can help me understand the configuration!

zeroby0, 2021-05-13:
1. Yes, you should use an Ethernet cable. WiFi will only be useable for collecting CSI after flashing Nexmon.
2. Any 11ac device should work in theory, but in practice, packets from some devices seem to not work. I have successfully collected packets transmitted from Archer C20, but cannot from C80.

I have [pre-compiled binaries and an install guide](https://github.com/nexmonster/nexmon_csi/discussions/2) that should make it easy for you to get started.

wolke1526, 2021-05-13:
@zeroby0 thank you for your kind reply and your guide work!