seemoo-lab/nexmon_csi #212: problem with install firmware, step 10

marioqwert9, 2021-05-13:
I have problem in step 10. How to fix this error?

BUILDING DRIVER for kernel 5.4 brcmfmac_5.4.y-nexmon/brcmfmac.ko (details: log/driver.log)
make[1]: *** /lib/modules/5.4.81-v7l+/build: No such file or directory.  Stop.
make: *** [Makefile:108: brcmfmac.ko] Error 2

zeroby0, 2021-05-14:
You don't have the right kernel headers.
You can get them as described here:
https://github.com/nexmonster/nexmon_csi/tree/pi-5.4.51-plus#get-kernel-headers

marioqwert9, 2021-05-19:
I install this kernel patch but now i have this error. How to fix?

[image]

marioqwert9, 2021-05-19:
I dont know is this the same error,how to fix ?
[image]

zeroby0, 2021-05-20:
Did you run `source setup_env.sh`? You have to do it everytime you are compiling Nexmon.

marioqwert9, 2021-05-20:
Yes i run this in nexmon directory and than make. Than i go nexmon_csi folder and run make install-firmware... I have install Kernel 5.4.50.

marioqwert9, 2021-05-20:
Can i record installation and post in on youtube. So you can tell me what i wrong do?

zeroby0, 2021-05-20:
You don't have to. I have a build guide that I have tested a few hundred times and I know works. https://github.com/nexmonster/nexmon_csi/tree/pi-5.4.51-plus

Follow it **exactly** on a freshly formatted sd card, and your installation will work. Connect to the Pi with an ethernet cable if you're using ssh from computer though.

marioqwert9, 2021-05-21:
Thank you, now i am on the last step with your tutorial. Tell me what you think with this Use ctrl-b d to detach, and tmux attach-session -t nexmon to re-attach.

zeroby0, 2021-05-21:
Tmux opens a new virtual terminal inside your terminal. To get out of the terminal, use ctrl + b and then press d. To enter the tmux terminal again, use `tmux a`. It'd be a useful skill and would save time to google things :) https://www.google.com/search?q=tmux

marioqwert9, 2021-05-21:
Ok, thank you very much. Is it need to to with new terminal, or not?