seemoo-lab/nexmon_csi #213: ASUS RT AC86U crash at high transmission rate

DennisHerell, 2021-05-17:
Hi, I've installed nexmon csi on asus rt ac86u and I managed to capture CSI data. However, when I set the transmitter rate to be higher it suddenly stop capturing CSI from my transmitter. It can still capture some CSI from other devices when I didn't set any MAC filter. (However, the capture rate become much slower). I also encounter this problem only when I am in place with high wireless activity and when I set the channel to 80 MHz. Does anyone else got this problem?

zeroby0, 2021-05-17:
Hey Dennis!

How many packets are being transmit per second when you set it to high? And are you sure the transmitter is not crashing? My Pi seems to abandon sending pings when I try to send more than 2000 pings per second.

And where are you saving your pcap file?

DennisHerell, 2021-05-18:
Hi @zeroby0, I'm setting at around 100 frames per second. I'm pretty sure it's not the transmitter, to test it I used both asus router and nexus 6P as CSI receiver, and the nexus 6P can capture the CSI just fine. In this scenario it crash even if I didn't specify to write the pcap files. (Although for 20 MHz channel, I can capture the data and save it to my laptop directly by using tcpdump through ssh:
`ssh admin@ROUTER_IP /jffs/tcpdump -ieth6 dst port 5500 -c2000 -vv -w- > csi.pcap`

What I'm really curious about is whether the frames from other AP affect the asus router even when I filter for my transmitter MAC address, as I mentioned it worked just fine in places with low wifi activity. If that's the case, it probably doesn't work in 80 MHz because it will interact with even more frames.

yujianyuanhaha, 2021-06-17:
@DennisHerell hi Dennis, (1) I suggest to save down in external USB disk, ASUS router support the USB storage locate in `/mnt/YOURDISK/`, ssh include file transfer delay, which may effect the timestamp of CSI if you care.  (2) it is true 100Hz is the highest I can get, mean nearly 1600 frame per second for all 16 spatial stream. It can be done by setting u-code using `nexutil -d` option.

Best

Jet