seemoo-lab/nexmon_csi #215: error occurs at step 7 " recipe for target 'gen/ucode.bin' failed "

wolke1526, 2021-05-18:
hi, I'm using asus ac86u router, when I follow the instruction to step 7 and run `make install-firmware REMOTEADDR=<my router's ip>` 
then I have this error
` PATCHING UCODE src/csi.ucode.bcm4366c0.10_10_122_20.patch => src/csi.ucode.bcm4366c0.10_10_122_20.asm
  ASSEMBLING UCODE src/csi.ucode.bcm4366c0.10_10_122_20.asm => gen/ucode.bin
Makefile:218: recipe for target 'gen/ucode.bin' failed
make: *** [gen/ucode.bin] Error 1`
I've already ran `source setup_env.sh` and `make` before under `sudo su`, can I know how to fix this? Thanks in advance!

zeroby0, 2021-05-18:
There should be log files in the folders nearby that will tell you what the error is in a more detailed manner.

You are also welcome to use my [precompiled binaries](https://github.com/nexmonster/nexmon_csi/discussions/2) and save time, or follow my [build guide](https://github.com/nexmonster/nexmon_csi/discussions/7).

wolke1526, 2021-05-20:
@zeroby0 Thank you for your advice! now I'm trying to use your precompiled package, but I have a problem and wonder if you can help me. when I try to send the package to my router,  it seems the connection cannot be established and it keeps reporting connection timeout, but I already have turned the ssh enable on the router website. Do you have any advice? much thanks!

wolke1526, 2021-05-20:
Thank you for your reply! I'm connect the router with a LAN cable. but after then I try use a usb flash to transfer the file to the router, and it work! and I have managed to capture some packets now, the only thing is I have to login in to the router's terminal, I'm sorry to disturb you but do you have any idea that I can save the pcap file to PC? thanks again for your work!

zeroby0, 2021-05-20:
Which operating system is your PC running?

wolke1526, 2021-05-21:
> Glad to hear that! Which operating system is your PC running?
> 
> Edit: It's probably not as fast as a flash drive, but you can use this: `ssh admin@ROUTER_IP /jffs/tcpdump -ieth6 dst port 5500 -c2000 -vv -w- > csi.pcap`. Credits to @DennisHerell , [#213 (comment)](https://github.com/seemoo-lab/nexmon_csi/issues/213#issuecomment-842983736)

It's running xubuntu 18.4.03 as suggested. I still have some problems with ssh connection but thanks for your advice!

wolke1526, 2021-05-21:
@zeroby0 It's running xubuntu 18.4.03 as suggested. I still have some problems with ssh connection so now I'm using `telnet ip` to login, and I have a question, the ac86u is monitoring the wifi packet over the air, is that mean that I need to use another device like a phone to communicate with the AP while the ac86u is listening? my configuration is, an xubuntu desktop connected with ac86u by a Ethernet cable though LAN port, and ac86u is connected with an other router by Ethernet cable though WAN port. I'm wondering if I get the right CSI data?

zeroby0, 2021-05-21:
What problems are you facing with SSH? Try pinging the ac86u and see if you have any packet drops. You can also use ssh -v to generate more debug info.

Yes, there should be some wireless traffic in the air to calculate CSI from.

I my configuration, I have a WiFi router connected to the Internet. Then I connect the WAN port of the ac86u to a LAN port on the WiFi router. Then I connect my laptop to the WiFi router via WiFi. 
Now I can SSH into the ac86u via the WiFi router. So the SSH connection going over WiFi also generates packets which the ac86u can collect CSI from.

sri-abhinav, 2023-01-25:
Hi @zeroby0 @wolke1526, I am facing similar issues.
1. In step 7, after running "make install-firmware REMOTEADDR=<address of your rt-ac86u>"
Makefile:221: recipe for target 'gen/ucode.bin' failed
make: *** [gen/ucode.bin] Error 1

2. In step 10 after running "scp nexutil admin@<address of your rt-ac86u>:/jffs/nexutil"
ssh: connect to host 192.XXX.XX.XX port 22: Connection refused
lost connection 

Could you please help me to fix this issue?

Thanks!

zeroby0, 2023-01-27:
I can't be of much help without the log files :D

As for 2. ssh connection failed. Is the IP correct and is ssh enabled on the router?