seemoo-lab/nexmon_csi #219: Flip on relative phase between antennas

Deltaaaaaaaaaa, 2021-06-01:
Thank you for this amazing CSI tool!
I'm collecting CSI using AC-86U as receiver(4 antennas) and Nexus6P as transmitter(1 antenna), to estimate AoA.
In a static environment, I found the relative phase between antennas filp over time, which makes AoA estimation unreliable.
This is a heat map of CSI phase with respect to the first antenna.
[image]
Have you guys met the same problem?
What might result in this flip? 
Is there some related signal processing procedure in AC-86U?
How can I calibrate this?
Thanks in advance!

Deltaaaaaaaaaa, 2021-06-03:
@quantumhub @yujianyuanhaha @matthiasseemoo Greetings! Could you guys give me some hint? Thanks in advance!

yujianyuanhaha, 2021-06-03:
@Deltaaaaaaaaaa  I am glad you look into the AoA or AOA-ToF estimation, I am met the same issue.

 Although tons of work is has been done but much of them is noisy  and hard to implement.

(1) Spoty [[paper link](https://web.stanford.edu/~skatti/pubs/sigcomm15-spotfi.pdf)]  [[open source link](https://bitbucket.org/mkotaru/spotfimusicaoaestimation/src/master/)], this work require you to swap the ant to calibrate the antenna.

(2) a more recent one [Robust and Practical WiFi Human Sensing
Using On-device Learning with a Domain Adaptive Model ](https://www.andrew.cmu.edu/user/esoltana/Elahe%20Soltan_files/papers/Buildsys20_MWiFi_CameraReady.pdf) argue diff over diff spatial stream should be constant over package, while such claim does not match with your observation.

Let me know if you figure out a better way

Best

Jet

Deltaaaaaaaaaa, 2021-06-04:
> @Deltaaaaaaaaaa I am glad you look into the AoA or AOA-ToF estimation, I am met the same issue.
> 
> You can easily unwrap the phase using pseduo code like below
> 
> ```
> if Phase(j) -  Phase(j-1) > pi
>         Phase(j) = Phase(j-1) - 2*pi;
>   elseif Phase(j)- Phase(j-1) < -pi
>          Phase(j) =Phase(j)+ 2*pi;
>  end
> ```
> 
> but it does not fixed the issue ....
> 
> Although tons of work is has been done but much of them is noisy and hard to implement.
> 
> (1) Spoty [[paper link](https://web.stanford.edu/~skatti/pubs/sigcomm15-spotfi.pdf)] [[open source link](https://bitbucket.org/mkotaru/spotfimusicaoaestimation/src/master/)], this work require you to swap the ant to calibrate the antenna.
> 
> (2) a more recent one [Robust and Practical WiFi Human Sensing Using On-device Learning with a Domain Adaptive Model ](https://www.andrew.cmu.edu/user/esoltana/Elahe%20Soltan_files/papers/Buildsys20_MWiFi_CameraReady.pdf) argue diff over diff spatial stream should be constant over package, while such claim does not match with your observation.
> 
> Let me know if you figure out a better way
> 
> Best
> 
> Jet

Hi, Thanks for your reply!
Have you tried to look into the frequency of the phase flip?
It seems to happen randomly, but the is added phase value is fixed.
I think research on the condition or the frequency of the flip might be helpful to calibrating it.

Best
Jimbo