seemoo-lab/nexmon_csi #221: Is carrier frequency Offset (CFO) removed before obtaining CSI?

Colin-Guolin, 2021-06-09:
Hey Guys,

I am using Rasp-pi to collect CSI, and I found the CSI pattern in time series is quite different after temperature changed, this may due to the carrier frequency offset changes. In standard 802.11 protocol, the CFO was estimated and compensated before the channel estimation, I am wondering in Nexmon CSI tool, is this procedure performed before obtaining the CSI?

matthiasseemoo, 2021-06-09:
The CSI is read when the Wi-Fi chip is already successfully receiving data. Hence, I assume that the CSI we measure is after CFO correction. Additionally, I assume that the CSI measurement might vary over time during a frame reception as pilot subcarriers are use to cope with effects such as doppler shift. However, I am not sure how this affects our CSI measurement. We only observed, that we are not able to successfully receive frames as soon as we start reading from the CSI registers. Hence, we wait until we fully received a frame and then read the CSI registers. Which, of course, reduces the CSI measurements per second. One way to figure out the effect of time would be to dump the received samples of a frame into the template RAM while receiving a frame and extract the CSI data and compare it to the value manually extracted from the captured LTF.

Colin-Guolin, 2021-06-09:
> The CSI is read when the Wi-Fi chip is already successfully receiving data. Hence, I assume that the CSI we measure is after CFO correction. Additionally, I assume that the CSI measurement might [...]

Thanks for replying, 

The information on the long training symbol would be valuable for me, could you give me some hint, how to extract raw training symbol data using Nexmon CSI?

matthiasseemoo, 2021-06-09:
> … <x-msg://12/#>

currently, only transmissions work. receptions are very experimental and not really understood yet: http://nexmon.org/sdr <http://nexmon.org/sdr>

yujianyuanhaha, 2021-06-11:
@Colin-Guolin I guess not, you can compare the phase diff between different packages/ different spatial streams/ different subcarriers, or in another way, cable connect 1 TX ant to another RX ant as to enforce direct path to see how phase diff is like.

Best

Jet

Colin-Guolin, 2021-06-11:
> @Colin-Guolin I guess not, you can compare the phase diff between different packages/ different spatial streams/ different subcarriers, or in another way, cable connect 1 TX ant to another RX ant as to enforce direct path to see how phase diff is like.
> 
> 
> 
> Best
> 
> 
> 
> Jet

Is this possible in raspberry pi?(connect two with cable)

yujianyuanhaha, 2021-06-11:
@Colin-Guolin I do not have rasberry pi but you can try that.

yujianyuanhaha, 2021-06-20:
@Colin-Guolin another work you can refer to is "Tewes, Simon, and Aydin Sezgin. "WS-WiFi: Wired Synchronization for CSI Extraction on COTS-WiFi-Transceivers." IEEE Internet of Things Journal 8.11 (2021): 9099-9108.", it mentioned the temperature and attennas imperfection that bring the offset.

elmehdiilli, 2021-08-15:
> The CSI is read when the Wi-Fi chip is already successfully receiving data. Hence, I assume that the CSI we measure is after CFO correction. Additionally, I assume that the CSI measurement might [...]

@matthiasseemoo Hello again dear Matthias,

So if I understood well from your reply, the CSI value captured manually from the LTF is a frequency-corrected one, and the samples stored in the template RAM are not corrected?

Many thanks once again,

vlsireddy, 2023-09-26:
Hi,
   so there need to be [CFO, Phase Offset, Sampling Time Offset] correction further . Is that summary of the discussion.

Thanks